Ultimaker/Cura #9143: Can't start Cura without printer

ColinPitrat, 2021-01-18:
I used to use an old version of Cura on my personal computer to prepare prints that I was then executing at my Fablab.
This was working well.
I recently updated my Cura version and found out that I cannot start it without connecting to a printer, which prevents me from preparing my print ...

Could you please add a "Skip" option to the screen for connecting to a printer?

fieldOfView, 2021-01-18:
See "Add a non-networked printer". This is not about connecting a printer, but telling Cura what sort of printer you want to slice for.

ColinPitrat, 2021-01-18:
This screen is completely empty for me apart from a field to name my printer:
[image]

fieldOfView, 2021-01-18:
Then we need to see your cura.log. It can be found in `%APPDATA%\cura\4.8\`.

ColinPitrat, 2021-01-18:
Sorry, I should have provided more details as it turns out it's not "work as designed".
I'm using v4.7.1 on ubuntu.
There's a $HOME/.config/cura/4.7 directory but no logs in it.
`strace cura 2>&1 | grep open | grep "\.log"` doesn't show any log file being opened (even `cura --debug`).

fvrmr, 2021-01-18:
On ubuntu you can find it here: `$USER/.local/share/cura/<Cura version>/cura.log`

fieldOfView, 2021-01-18:
Given that you are not using the current version, I suggest you also try the official AppImage provided by Ultimaker.

ColinPitrat, 2021-01-18:
`~/.local/share/cura/4.7/` exists but doesn't contain any log:

```
$ find ~/.local/share/cura/ -name \*.log
$ 
```

The AppImage version works fine though, so I'll go with it.

Thanks for your help. I'll closed this for now as this solves the problem on my side.

closed 2021-01-18 by ColinPitrat