Ultimaker/Cura #9144: [4.8.0] Cura fails to start slicing

KaraokeAmerica, 2021-01-18:
**Application version**
4.8.0

**Platform**
Windows10, Intel Core-i7 CPU.

**Printer**
Tried both Ultimaker 3 and Ender3

**Reproduction steps**
1) Imported STL
2) Try to slice using default or modified settings

**Screenshot(s)**
[image]

**Actual results**
Nothing. It just sits there at 0% and never progresses. Program is not hung per say. I can close it and it responds to input. It just never slices.

**Expected results**
Model would slice is what I would expect, and it is what happened until a few days ago without making any changes in Cura.

**Project file**
Two files are in here. I printed one of them about 10 or so days ago without issue from this same system after slicing it in the same install of Cura 4.8.0. The other was one I was trying tpo slice when I first saw this issue.
[Project Files.zip](https://github.com/Ultimaker/Cura/files/5832323/Project.Files.zip)

**Log file**
This is from trying to slice the GlueGunStand.3mf project.
[cura.log](https://github.com/Ultimaker/Cura/files/5832329/cura.log)

**Additional information**
I have completely uninstalled all versions of Cura and deleted the appdata folders for all versions as well to get a fresh install on the system. Still didn't resolve the issue. I then tried the new alpha version with the same result. I have also created a new profile to see if anything was corrupt. Currently I have the alpha and v4.8.0 still on the system.

nallath, 2021-01-19:
"2021-01-10 10:13:13,996 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] [ERROR] Could not connect to the given address"

Do you have a firewall enabled? Something is blocking the (local) connection between the GUI and the slicing engine.

KaraokeAmerica, 2021-01-19:
> Do you have a firewall enabled?

So.......no firewall other than the built in Windows Firewall that has always been there. Nothing new. To be sure that wasn't the issue I went and deleted the 8 cura entries the then recreated a manual entry for the cura.exe in the v4.8 directory. No change.

However, I also run a VPN: Nord. They released a new version that includes a new, split-tunneling feature. I did some trials....
VPN disconnected: Cura slices
VPN connected with split tunneling feature off: Cura slices
VPN connected with split-tunneling on and Cura added to EXCEPTION list: Cura does NOT slice
VPN connected with split-tunneling but browser only added to INCLUSION list: Cura does NOT slice. 

Windows firewall doesn't seem to have any effect.

Conclusion: Cura doesn't work with NordVPN split-tunneling enabled in any way. That's messed up. I would think that it would work with Cura being added to an exception list assuming it works properly. However, it begs the question of why Cura requires a network path to run locally? Is it connecting to the internet to slice something? If not, what process is being blocked exactly? Perhaps I can add that to the exceptions list to see if ti works with VPN enabled.

fvrmr, 2021-01-22:
Thanks, we will discuss this with the team. Keep you posted!

rburema, 2021-01-22:
>  However, it begs the question of why Cura requires a path that can be blocked by a VPN to run locally to begin with? Is it connecting to the internet to slice something? If not, what process needs access exactly? Perhaps I can add that to the exceptions list to see if ti works with VPN enabled.

We make a connection to the slicing engine running locally (so, from your computer, to your computer). Localhost is 127.0.0.1 It's a bit strange that a VPN blocks local traffic like that?

fvrmr, 2021-01-25:
It looks like this is a problem with NordVPN and not Cura. Unfortunately we can't help you with this. But you can find more information here:
https://www.reddit.com/r/nordvpn/comments/kqrq1a/nordvpn_split_tunnelling_and_localhost/ and https://stackoverflow.com/questions/32103885/not-able-to-access-local-server-running-after-vpn-connection

Hopefully you can find your answer there.
So I am closing this issue because it is not Cura related.

closed 2021-01-25 by fvrmr