Ultimaker/Cura #9145: Preview

longlivekel, 2021-01-18:
**Application version**
(The version of the application this issue occurs with.)
4.8

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Mac Big Sur 11.1

**Printer**
(Which printer was selected in Cura?)
Creality CR10

**Reproduction steps**
Slice/save file Then see the preview. Preview is blank

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)
Preview to work

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)
3dbenchy file

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

ehusar, 2021-01-19:
I can confirm the same issue. I am also on Big Sir 11.1.  There is a temporary fix that has been working for me.  Go to Preferences / General.  Check the box for Force Layer View Compatibility (restart required).

longlivekel, 2021-01-19:
Do i have to do this every time?
I am glad that 4.8 isn't crashing. About a week ago it would crash non
stop. I got tired of 4.7 poor print quality so I installed 4.8 this
morning. Between 4.7 and 4.8 (when it worked) both would not do a preview.
I was forced to go to 4.7.X due to the crashing of 4.8, but still 4.7.x
would not do the preview.

fvrmr, 2021-01-19:
We know that we are having a lot of issues with Big Sur. And since this is a duplicate of https://github.com/Ultimaker/Cura/issues/8745 I am closing this issue.

closed 2021-01-19 by fvrmr