Ultimaker/Cura #9146: Not possible to clean the build plate

bmwiX, 2021-01-19:
**Application version**4.8.0.
(The version of the application this issue occurs with.)

**Platform**Mac OS 11.2
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer** Ultimaker 3
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
loading a model2. (Something you did next.)trying to clear the plate

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results** nothing
(What happens after the above steps have been followed.)

**Expected results** a clear plate ready for loading
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-01-19:
Can you share your log file? You can find that here: $USER/Library/Application Support/cura/<Cura version>/cura.log

bmwiX, 2021-01-19:
> Can you share your log file? You can find that here: $USER/Library/Application Support/cura//cura.log

Thanks, but it is not possible to find  cura file under application support. Nevertheless I found it under : help/show configuration folder/cura. After deleting all cura files there,  Cura works well again.

Regards

Morten Dragland
Bjørnefjellet 6 
1613 Fredrikstad

mdragland@mac.com
mob.  91381565

jellespijker, 2021-01-19:
@bmwiX good to hear that it solved your problem. I will now close this issue.

closed 2021-01-19 by jellespijker