Ultimaker/Cura #9148: Change Extruder parameters

jornada812, 2021-01-19:
To be honest, this is no longer funny. For a hundred years now, saving values for the extruder has not worked. They wrote that it was once corrected, then it comes back again. Neither the nozzle diameter nor the filament thickness can be changed. This is not saved in the config file! Programmers! Is it so difficult to check for the existence of a section in the config and add parameters there? This is nonsense! What version lasts already!

fieldOfView, 2021-01-19:
This is an issue queue, not a rant queue. If you have an issue, please fill out the issue template. As is, I have no idea what your issue is, so there is nothing I can do to help you.

jellespijker, 2021-01-19:
Please fill in the issue template and add the requested files.

closed 2021-02-02 by no-response[bot]