Ultimaker/Cura #9151: [4.8] Top surface skin layers active while top layers are set to 0

printingotb, 2021-01-20:
**Application version**
4.8

**Printer**
Custom fff

**Reproduction steps**
1. Enable 1 top surface layer with concentric pattern. Set bottom layers to 1 with lines pattern, set top layers to 0.
This uses the top surface skin layer settings for the only layer left which is a bottom layer.
2. Re enabling top layers and setting top surface layers to 0 and then top layers to 0 will print the bottom layer with it's own settings.

**Screenshot(s)**
[image]
[image]

**Actual results**
As described above

**Expected results**
Top surface skin layers should disable with the top layers set to 0.

**Project file**
[Box with walls.zip](https://github.com/Ultimaker/Cura/files/5841792/Box.with.walls.zip)

fvrmr, 2021-01-22:
Can you give us more information, because I see that your object has only one layer, so that layer will be the top layer.
And when I scale your object, the bottom layer is with lines, and the top layer is concentric.

printingotb, 2021-01-25:
Sorry, didn't see your answer before. 

I set the top surface skin pattern to concentric to be able to easily tell "top surface skin layers" (from the setting with the same name) apart from normal "top layers" with lines pattern.

I made a new simpler model with 10mm thickness. 0 Top layers and 999 bottom layers to completely fill it with bottom layers.
Number of "top surface skin layers" (the setting) was set to 1 before reducing the normal top layers to 0. When doing this, Cura hides the number of top surface skin layers setting and it's related settings, so one would assume that the top surface skin specific settings (concentric line pattern, line width, speed etc) would not be in effect anymore.

When number of top layers are set to 0, it seems like Cura is treating the topmost "bottom layer" as a "top layer", and since the number of top surface skin layers are set to 1, it applies top surface settings to it. 

- Is Cura supposed to treat the topmost bottom layer as a top layer even though top layers are set to 0? 
Should it still apply "top surface skin" settings to it?

Thanks!

[Top surface.zip](https://github.com/Ultimaker/Cura/files/5866610/Top.surface.zip)

[image]
[image]
Both these is slicing the model with the concentric "top surface skin layer".

Ghostkeeper, 2021-01-29:
Right, I understand what's happening, yeah.

Internally, the top and bottom are just a single area per layer. It's not kept separate. This is why the Top/Bottom Pattern, Speed, Line Width, etc. settings can't be changed separately for the top and the bottom; there's just one area to fill with that pattern. It's desirable to merge these together, or you'd get a separation line between the two areas if any of its settings are different and this is not what you'd want in your print.

So the Top Surface Skin Layers are taken from those combined areas, not just from the part of it that was originally "bottom".

The front-end will hide the Top Surface Skin Layers setting if the Top Layers setting is 0, assuming it doesn't apply then. But it actually does apply, and this is incongruent. So we have two options here:
* Make the Top Surface Skin Layers setting (and related pattern, speed, etc) visible even if Top Layers is 0. It could depend on Top+Bottom layers being more than 0 or something. Or...
* Limit the Top Surface Skin Layers setting to be always less than the Top Layers setting in CuraEngine, thus applying it only to the Top Layers.

I would be in favour of implementing the limit in CuraEngine since that's more like what I would expect. It's also what Printingotb was expecting.
However we do need to consider the case of 100% infill. Instead of setting the Bottom Layers to 999999, we might want to set the Top Layers to that value then.

closed 2021-02-01 by Ghostkeeper

Ghostkeeper, 2021-02-01:
We've limited the Top Surface Skin Layers setting in CuraEngine. So the setting then still gets hidden but it then actually doesn't influence the bottom layers any more, as you'd expect from a hidden setting.

printingotb, 2021-02-01:
Nice :+1: