Ultimaker/Cura #9154: Make "Line type" colour schme the default for previewing

printingotb, 2021-01-20:
**Is your feature request related to a problem? Please describe.**
It is quite common that new users and some not very new users hasn't discovered the colour schemes. It would shorten the learning curve if _line type_ was the default.

**Describe the solution you'd like**
Make line type colour scheme the default.

**Describe alternatives you've considered**
Perhaps only make it default if the machine profile isn't using multiple extruders.

**Affected users and/or printers**
Mainly beginners but also people who help others troubleshoot print issues.

fvrmr, 2021-01-21:
Thanks, will discuss this with the team. Keep you posted!

Ghostkeeper, 2021-01-22:
This request prompted us to look into this default again. All of our engineers know Line Type to be the best one. And also in the field it's the one we see all of the screenshots are made with, which is a good indicator that most users will have been using that one (so they did discover it, luckily).
The original reasons to use Material Colour are less relevant now, so we've decided to change this and notify our UX developer of it (he wasn't present at the time we discussed it).

closed 2021-01-22 by Ghostkeeper

printingotb, 2021-01-22:
Excellent, thank you!

commit f28b90a1 (2021-01-22): Make Line Type the default selection for layer view colour scheme

Implements CURA-7983.
Fixes #9154.