Ultimaker/Cura #9155: Z-distance as number of layers

printingotb, 2021-01-20:
**Is your feature request related to a problem? Please describe.**
It is fairly common for users to think that they can set support Z-distance to any number they want and have that distance instead of the rounded up distance used by Cura. Comments like "decrease Z-distance by 0.02mm and try again" or "I always use a Z-distance of 0.24 for 0.2mm layer height" are sometimes given as advise.

**Describe the solution you'd like**
Just like number of walls and top/bottom layers can be set both as millimetres and numbers, so could support Z-distance.
This would make it more obvious that the mm value isn't exact.
I get that a chosen value in millimetres could be tweaked to automatically round to the needed number of layers gap based on the layer height but it's not perfect and you end up changing it manually anyway.

**Additional context**
[image]

Ideally I would want a dropdown to choose the amount of layers the gap should be. This makes the most sense if another of my feature requests would be implemented in the future :wink: #9019 continued here: #6105

fvrmr, 2021-01-21:
Thanks. I will also discuss this with the team. Keep you posted!

fvrmr, 2021-01-22:
We will defer this. It is not on our priority list for now.

MariMakes, 2023-11-16:
Quick update on our side, you can now enter values that deviate from your layer height for the Support Z Distance. 
Can you please try it out and let us know if it resolves your issue?
You can download it here. https://github.com/Ultimaker/Cura/releases/tag/5.6.0-beta.1

closed 2023-11-30 by github-actions[bot]