Ultimaker/Cura #9156: Problems with opening project files (.3mf)

Cotwoman, 2021-01-20:
I want to open my projects as projects and not only import the models. Default behavior is set to "always ask". But with some project files the dialogue window does not even appear and when the default behavior is set to "always open as projects" it still only imports the 3D-model, not the print settings. All my project files are in the .3mf format. I already reinstalled Cura, it's the newest version, feel like Curas Project safe-/import-process is buggy.

fvrmr, 2021-01-21:
Can you share some of your project files, that will not import as a project file? 

And for next time please fill in the issue template and add the requested files.

Cotwoman, 2021-01-22:
Sure.
[UMS5_Stempel_PVDF (2).zip](https://github.com/Ultimaker/Cura/files/5854662/UMS5_Stempel_PVDF.2.zip)

fvrmr, 2021-01-22:
Thanks for your file. I can't open this as a project file too.
Do you use `export` for saving project files?

Cotwoman, 2021-01-22:
Yes, that could be the problem.

fvrmr, 2021-01-22:
Ah yeah, that should be the problem then. To save a profile use `save project`

Let me know if it is not working. I will close this issue, because I think we found it.

closed 2021-01-22 by fvrmr