Ultimaker/Cura #9157: Infill before walls not applied when infill density set to 100%

tommylux, 2021-01-20:
**Application version**
4.8

**Platform**
Windows 10

**Printer**
Geeetech A20T

**Reproduction steps**
1. Ensure Infill before Walls is selected, note the nozzle path in Preview fills in infill before outer walls.
2. Set the Infill density to 100% note the infill is now applied last.
3. Adjust the infill density to 99% and note the infill is now applied first.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]
[image]

**Actual results**
Infill is applied last despite the checkbox infill before walls is checked.

**Expected results**
Infill before walls should work despite infill density stting

**Additional information**
I have tired enabling and disabling travel optimization.

fvrmr, 2021-01-21:
If you set the `top layers` and `bottom layers` on 2 (or something else). You'll see that after the bottom layers the infill is printed first. 

[image]

jellespijker, 2021-01-21:
Hi @tommylux ,

Thank you for bug report.

You're indeed correct that Cura doesn't print infill before walls, when infill is set a 100%. That is because we emulate the infill by replacing it with bottom skin. @Ghostkeeper describes why this approach is taken in the following discussion thread https://github.com/Ultimaker/Cura/discussions/8965#discussioncomment-235730

> In Cura, if you set the infill density to 100%, and inspect layer view, you'll find that it didn't actually create infill. That's because it then automatically sets the Bottom Thickness to 9999mm. This eliminates a border between skin and infill where the lines don't match up. You only have walls+skin then. In the end this results in better print quality and fewer air pockets.

As a side effect certain settings, such as infill before walls, don't work as expected anymore. A work around for this can be to set the `extra infil wall line count` to a higher number like `9999`. This would create a 100% infill filled volume, which adheres to the printing order.

I will discuss it with my fellow team members if/when we will adres this. Addressing this behaviour in the new Arachne seems like a logical point to me since Arachne takes care in filling the complete area, without air pockets. Which might occur if you use the proposed work-around.

fvrmr, 2021-01-22:
We will create a ticket to investigate this with our process engineers. 

If you look at my work around, the infill is not completely 100% if you change it to grid. So you need to add what @jellespijker described above

> As a side effect certain settings, such as infill before walls, don't work as expected anymore. A work around for this can be to set the `extra infil wall line count` to a higher number like `9999`. This would create a 100% infill filled volume, which adheres to the printing order.

tommylux, 2021-01-22:
Thanks. In fact, I don't particularly mind which is applied first. Only, that I was using the wall to slow down the print speed before it travels to the next place to reduce stringing. I think the better parameter to use is reduce the coasting speed <90% and increase coasted volume. Is there a preference generally within the community to print walls first?

tommylux, 2021-01-22:
Fantastic explanation from jellespijker and this makes complete sense and a good insight to how Cura is programmed.

fvrmr, 2021-01-22:
Developers see CURA-7984

closed 2024-12-08 by github-actions[bot]