Ultimaker/Cura #9159: Cura slicer won't slice while vpn active

asdfmonster261, 2021-01-21:
**Application version**
4.8.0

**Platform**
Windows 10 x64 build 19041.746, GTX 1070 TI, Intel i5-8400

**Printer**
Creality Ender 3 Pro, Octoprint

**Reproduction steps**
1. Turn on NordVPN with split filter, Cura is not in the filter to go through VPN.
2. Load stl in Cura
3. Press slice
4. Wait, progress bar does not move
5. Cancel slice
6. Disable VPN
7. Press slice, slicing now works

**Screenshot(s)**
N/A it's just the slice button

**Actual results**
Slicing does not begin, by my understanding Cura connects to curaengine on a specific port on localhost, this is not happening.

**Expected results**
Cura to slice with VPN active

**Project file**
Project file is irrelevant, happens on any file with VPN open.

**Log file**
https://pastebin.com/YjhL047R

nallath, 2021-01-21:
You're right: 
```
2021-01-21 01:16:00,456 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [110]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 4.8.0\\CuraEngine.exe', 'connect', '127.0.0.1:49675', '']
2021-01-21 01:16:00,534 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] [ERROR] Could not connect to the given address

```
So it seems that the local communication is being routed through your VPN.

fvrmr, 2021-01-22:
I will close this issue, because it is an duplicate of #9144

closed 2021-01-22 by fvrmr