Ultimaker/Cura #9163: Custom Prefix for Custom FFF Printer on saved GCODE File

Carl1509, 2021-01-21:
Hey,
im not sure if this is the right format, but i give it a try.
I have a few custom build 3d printers. With preconfigured printers, cura does a great job, with giving each file a custom prefix, so you wont put the wrong sliced file in a wrong printer.
With custom FFF Printer, the prefix is always "CFFP". With many printers, its quite chaotic for me.
It would be nice if there is a way to create a custom prefix for each printer, without modifying each file.

**Application version**
4.8

**Platform**
MacOS

**Printer**
CustomFFF Printer

**Reproduction steps**
1. Create a custom FFF Printer Machine
2. Load an STL
3. Slice it
4. "Save as File"

**Actual results**
Output file is something like CFFFP_'slicedSTL'.gcode

**Expected results**
Output file is something like 'youCustomModifier'slicedSTL'.gcode

fieldOfView, 2021-01-21:
In the Marketplace, there are two plugins that let you do this.
[CustomJobPrefix](https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/CustomJobPrefix) and [GcodeFilenameFormat](https://marketplace.ultimaker.com/app/cura/plugins/rgomezjnr/GcodeFilenameFormat). I am partial to the former, since I created it.

fvrmr, 2021-01-22:
Thanks @fieldOfView 
You can indeed use these plugins. 

Because there is an solution for this I will close this issue.

closed 2021-01-22 by fvrmr