Ultimaker/Cura #9165: [4.8.0] Raft, brim, skirt don't respect first layer print & travel speed

globules-io, 2021-01-21:
**Application version**
4.8.0

**Platform**
Windows 10

**Actual results**
Raft, brim, skirt printed at default speed

**Expected results**
Raft, brim, skirt should use first layer speed & travel

fvrmr, 2021-01-22:
If have tried it my self and the brim and skirt has the same speed as the initial layer.
The raft however not, because you print your object on a raft. 
So the question is do you want the initial layer or the first layer of your object the same speed.

I will also discuss this with the team. I'll let you know when I have more info about this.

globules-io, 2021-01-22:
The skirt here does not print at the same speed as the first layer. I just tested it as well ?

Ghostkeeper, 2021-01-25:
The skirt/brim and raft have their own speed settings.

By default (for most profiles at least), the speed for skirt and brim is set equal to the initial layer speed. But this doesn't necessarily have to stay the same. If you change the skirt/brim speed then it's not the same any more.

Since the Skirt/Brim Speed and the Raft Base Speed settings only apply to the initial layer, they wouldn't have any effect otherwise. So this is expected behaviour. I don't know which profile you're using (since you erased that from the issue template), but I'll assume that you changed the speed setting to a fixed value, or your profile does.

Ghostkeeper, 2021-01-25:
We've added some clarification to the setting description of the Initial Layer Speed to reduce confusion.

closed 2021-01-25 by Ghostkeeper