Ultimaker/Cura #9166: 4.8 Cannot see imported print profiles

RetArsonFD, 2021-01-22:
**Application version**
(The version of the application this issue occurs with.)
4.8, with the "Materials Settings" plugin

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10...64 bit

**Printer**
(Which printer was selected in Cura?)
Ender 3

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)
I attempted to import several print profiles into the "print settings", but I cannot see the profiles on my page.  I keep getting messages saying "The profile is not visible in the current configuration.  However, I have changed every configuration I could find on the page, and nothing made my profile visible to me.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
(What happens after the above steps have been followed.)
Nothing...as noted above

**Expected results**
(What should happen after the above steps have been followed.)
I expect to see my imported print profiles
[cura Bug.docx](https://github.com/Ultimaker/Cura/files/5852979/cura.Bug.docx)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-01-22:
Can yo share your log files?

And I am wondering if the printer profiles are really for the Ender-3.

RetArsonFD, 2021-01-23:
If you tell me how to find & share the log files, I'll be glad too.

Here are a few of the profiles I was trying to import

RetArsonFD, 2021-01-23:
Hi there.  Not sure if you'll get this email, but thank you  for the help.  I have no idea how to share the log files, but if you tell me how, I'm certainly willing to try.  Here are 3 of the profiles I was trying to import into 4.8.  GITHub would not let me attach them to my response.  All 3 say they are a "CURAPROFILE" type file.  Not sure what that means, but I would think it would be compatible with Cura.  Also, if it matters, I'm using eSun PLA+ as my filament.

THANKS again !!!!!
Tommy

fvrmr, 2021-01-25:
Hi @RetArsonFD I can't see your profiles because it is an e-mail. 
You can add them in a comment on Github as a .zip or share a link from Google drive etc. 

Your log file can be find here: C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log

Please let me know if it won't work.

RetArsonFD, 2021-01-26:
Hey there,

Thanks for getting back to me.  This link should take you to the file in my Google drive with the info you asked for (I think).  If it doesn't work, or you need more info, let me know.  For what it's worth, there are several profiles in the folder.  I only specified 3 in my original issue, but they all react the same when I try to import them to Cura.

Thanks again for the help.
Tommy

https://drive.google.com/drive/folders/1_AviGPvAFS3YuXDNVi-s907B8SQ2wA8v?usp=sharing

Your log file can be find here: C:\Users\\AppData\Roaming\cura<Cura version>\cura.log

fvrmr, 2021-01-26:
It looks like you can't import draft because there is no draft quality type in your default settings. I tried to import `CHEPV4.5_0.12(best).curaprofile` and I can import that one because the `quality_type` = super. And that is supported with your current settings. Draft is not available for your printer. 

I hope this makes sense to you. Please let me know if you need any help or have more questions.

RetArsonFD, 2021-01-26:
Thanks.......but it doesn't make sense to me that it "isn't available to my printer".  How do I fix that ??  I got that list of profiles from someone on an Ender 3 Facebook page that claims to run those profiles on his Ender 3,  and highly recommended the "Draft" profile.

Thanks again for the help.

Sent from my Verizon, Samsung Galaxy smartphone

fvrmr, 2021-01-29:
Hi @RetArsonFD 

You can find more information about importing profiles here: #8229 
In this thread you can find also a work around: 

> The easy workaround is to change the material type to something that has supported quality profiles.
> 
> Cura can't migrate these profiles automatically because it's missing information. In particular, these profiles don't have the information as to which built-in quality profile they were based upon. In Cura 4.6, we were not filtering custom profiles correctly, which made custom profiles appear regardless of what built-in profile they were based on. So even though a profile was based on the "draft" profile, it would appear in the list of profiles even if you're using a small nozzle size and there is no "draft" profile for that nozzle size. It would select an arbitrary profile then to base it on, which causes the print to be completely different.
> Now in 4.7, this filtering is fixed. The profile no longer appears in the list if it's based on a quality level that's not available. But if you were using the profile in that incomplete state it causes a workflow issue. The profile still stores that it's based on the "draft" profile, so it won't appear in the list if there is no built-in "draft" profile.
> 
> You could repair the old profiles manually in your configuration folder, which would cause the profile to finally become stable, actively marking it as being based on an empty quality profile. This involves making changes to the configuration folder though, so it's easy to make mistakes and you should make a back-up first.
> What you would need to do is to go to Help -> Show configuration folder and close down Cura. Then search for the `quality_changes` subfolder and in there find all files that have a name similar to the profile you want to edit. There should be one file for each extruder that your printer has, and one extra file for the global settings. In each of these files, under the `[metadata]` tag, search for the `quality_type` line and change its value into `not_supported`. Save the files, then start up Cura. They will no longer be available for printer configurations that have supported quality levels, but they will then be available for those that don't.

I will close this issue, hope you can make it work with the work around. For more questions about this topic, please use #8229 so we keep everything together.

closed 2021-01-29 by fvrmr