Ultimaker/Cura #9167: FR: Toggle to allow Ignoring nozzle XY offsets when generating GCODE.

RFBomb, 2021-01-22:
**Is your feature request related to a problem? Please describe.**

I set up my nozzle offsets in my firmware so that T0 and T1 will move the print head to the correct location. 
I also set up my nozzle offsets in CURA, so that I could clearly see the printable area for each nozzle (thats a great feature by the way). 

Unfortunately, having the nozzle offsets described in both locations results in the printer missing the mark when moving between nozzles. For example:
 - my Nozzle 2 X offset is -24. 
 - T1 occurs, which tells the printer to move 24mm relative to nozzle 1 so that the active print head is in correct location for the gcode. 
 - Cura also offsets by 24mm, resulting in all T1 extrusion being off by 24mm from expected position. 

**Describe the solution you'd like**
Include a checkbox in the extruder settings that enables including the offsets into the gcode calculation by default , but unchecking the box will allow it to ignore them. 

This would allow the user to be able to see what the printable area is for each nozzle, handle it the offset in cura if the firmware doesn't already have the tool switching set up (which is how cura is currently set up), but allow firmware controlled tool switching to control that offset if desired. This can likely be accomplished by setting up a run-time variable for each nozzle when the script to run gcode is created. 

IF [Checkbox = true] {
   Nozzle_2_Offset X = User-Set Value
   Nozzle_2_Offset Y = User-Set Value
} ELSE {
   Nozzle_2_Offset X = 0
   Nozzle_2_Offset Y = 0
}

**Describe alternatives you've considered**
The workaround for this is obviously just set the cura offsets to 0, which is what I will end up doing.

**Affected users and/or printers**
Anyone that has a chimera-style printer could benefit from this. 

**Additional context**

Expected:
[image]

Actual (with offsets set in cura + firmware):

[image]

fvrmr, 2021-01-22:
I will discuss this with the team. Keep you posted!

RFBomb, 2021-01-23:
@fvrmr  -> This might already be implemented in the configuration files, it just isnt user-facing. 
While I was digging through the FDMPrinter.def.json file, I found this snippet:
```
                "machine_use_extruder_offset_to_offset_coords":
                {
                    "label": "Offset with Extruder",
                    "description": "Apply the extruder offset to the coordinate system.",
                    "type": "bool",
                    "default_value": true,
                    "settable_per_mesh": false,
                    "settable_per_extruder": false,
                    "settable_per_meshgroup": false
                },
```
If this is true, it would just mean exposing that option in the UI

fvrmr, 2021-01-25:
Thanks for your comment, but we will not implement this in the UI. It is not one of our priorities right now.
The right way to do this is set the offset in Cura to zero and shrink the build plate volume accordingly. 
I will close this issue, but please let me know if you have more questions.

closed 2021-01-25 by fvrmr

RFBomb, 2021-01-25:
I understand it may not be a priority, but I disagree that that is the
correct method to recommend.

Your suggested method has multiple issues.

- let’s say nozzle 1 can reach the whole thing, while nozzle 2 cannot.
Shrinking the build plate will cause nozzle 1 to have a limited print area.
Meaning you are without a doubt losing build area.

- If neither can reach full area due to the design, then it still isn’t
ideal. Say T0 can reach the left, but not right. And T1 can right a strip
on the right, but not left. Those areas can still be used as purge zones or
support or whatever.

- The important one though is shrinking the build area will prevent the
printer printing at the center of the build plate. If your plate is
230x230, but is set to 200x230, then everything will always be 15mm shifted
on the plate. This will screw up any sort of Bed mesh people have unless
they account for it in their firmware as well.

As I said, I understand it not being a priority. But if the logic is
already implemented, which the fdmprinter.def.Json file suggests, it should
be fairly easy to add this in as a checkbox on the main printer page and
save it to the custom printer overrides (which are already being modified
when someone updates their start gcode for example). It doesn’t have to be
per-extruder.

RFBomb, 2021-01-25:
After digging, it appears that it is implemented within the ' CuraEngine/src/gcodeExport.cpp' file. 

I think that the proper recomendation here would have been telling someone to implement it in their printer's configuration file, as opposed to messing with their bed size and likely getting issues I noted above. Since I now have confirmation that it is indeed implemented, I will be applying this fix to my custom printer's file. 

As I said, I understand it not being a priority, and that most users would likely never need to use it (since most are just single nozzle systems), I get the hesitance to bring it out to the user. But my reply to that would be that until you start printing in such a setup, the issue is completely blind to the user. So if they run into the issue, like I did, then they could have a hard time figuring out what went wrong and how to correct it. I only knew what went wrong so easily because I had already been digging through the setup files for both Cura and Marlin. Exposing the option in the UI as a simple checkbox would allow them to immediately check/uncheck it and perform another test print if the issue did occur to them. And since, everything but the checkbox itself is set up, that is why I was pushing for this change.

RFBomb, 2021-01-25:
Took a little over an hour for me to find the file (without knowing what I was looking for), and about 5 minutes to modify. I'll test it in my Cura setup tonight. 
Proposed solution:
https://github.com/RFBomb/Cura/blob/master/plugins/MachineSettingsAction/MachineSettingsPrinterTab.qml

fvrmr, 2021-01-26:
Hi @RFBomb nice that you figured it out. 
A PR is always welcome, our developers will look into it!

RFBomb, 2021-01-26:
Ok, I’ll submit a PR for it then later in the week.

Which branch should the PR go to?

fieldOfView, 2021-01-26:
PRs should target master