Ultimaker/Cura #9168: [4.8.0] Crash while adding a new material

JoshMcCullough, 2021-01-22:
**Application version**

4.8.0

**Platform**

- Linux-5.10.1-2-MANJARO-x86_64-with-glibc2.32
- Qt: 5.15.2
- PyQt: 5.15.2
- OpenGL 4.6
- AMD Radeon RX 580 Series

**Printer**

Ender 3 Pro

**Reproduction steps**

1. slick material drop-down at top
2. click to add a new material
3. "Discard or Keep Changes" dialog displays (optional)
4. wait 2-3 seconds
5. application crashes

**Screenshot(s)**

[image]

**Actual results**

While the "Discard or Keep Changes" dialog is up, the application will crash. Not immediately, but 2-3 seconds after the dialog is shown. Note that the issue persists even if there are no changes to the current material (so the dialog doesn't show), or if you close/complete the dialog. Crash persists 2-3 seconds after clicking to add a new material.

**Expected results**

Shouldn't crash. :-) 

**Log file**

Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/cura/AutoSave.py", line 61, in _onTimeout
    self._application.saveSettings()
  File "/usr/lib/python3.9/site-packages/cura/CuraApplication.py", line 723, in saveSettings
    ContainerRegistry.getInstance().saveDirtyContainers()
  File "/usr/lib/python3.9/site-packages/cura/Settings/CuraContainerRegistry.py", line 487, in saveDirtyContainers
    self.saveContainer(instance)
  File "/usr/lib/python3.9/site-packages/UM/Settings/ContainerRegistry.py", line 604, in saveContainer
    container.setDirty(False)
  File "/usr/lib/cura/plugins/XmlMaterialProfile/XmlMaterialProfile.py", line 126, in setDirty
    if base_file is not None and base_file != self.getId() and not registry.isReadOnly(base_file):
  File "/usr/lib/python3.9/site-packages/UM/Settings/ContainerRegistry.py", line 321, in isReadOnly
    return provider.isReadOnly(container_id)
  File "/usr/lib/uranium/plugins/LocalContainerProvider/LocalContainerProvider.py", line 188, in isReadOnly
    file_path = self._id_to_path[container_id]  # If KeyError: We don't know this ID.
KeyError: 'custom_material'

Thread 0x00007f2c1bfff640 (most recent call first):
  File "/usr/lib/cura/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f2c30923640 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f2c31de5640 (most recent call first):
  File "/usr/lib/python3.9/site-packages/UM/Backend/Backend.py", line 165 in _storeStderrToLogThread
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f2c325e6640 (most recent call first):
  File "/usr/lib/python3.9/site-packages/UM/Backend/Backend.py", line 153 in _storeOutputToLogThread
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f2c4f7fe640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f2c4ffff640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f2c70ff9640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f2c717fa640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f2c71ffb640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f2c727fc640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f2c72ffd640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f2c737fe640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Current thread 0x00007f2cd6d9d740 (most recent call first):
  File "/usr/lib/python3.9/site-packages/cura/CrashHandler.py", line 396 in _logInfoWidget
  File "/usr/lib/python3.9/site-packages/cura/CrashHandler.py", line 182 in _createDialog
  File "/usr/lib/python3.9/site-packages/cura/CrashHandler.py", line 86 in __init__
  File "/usr/bin/cura", line 167 in exceptHook
  File "/usr/lib/python3.9/site-packages/UM/Qt/QtApplication.py", line 378 in exec_
  File "/usr/lib/python3.9/site-packages/cura/CuraApplication.py", line 847 in run
  File "/usr/bin/cura", line 235 in <module>

forenbuddy, 2021-01-23:
Seems to be the same problem as here: https://github.com/Ultimaker/Cura/issues/9039 reported nearly a month ago.

Same problem holds up Arch Linux and Debian users of Cura 4.8 except of course they use the Appimage.

Also Fedora is wating for it as it looks:
https://bugzilla.redhat.com/show_bug.cgi?id=1912065

The only answer people seem to get is to use the Appimage which is a binary file. Who knows what is different in the Appimage.

So strictly speaking a working Cura right now is not open source as it looks to me.

I hope this changes again soon.

fvrmr, 2021-01-25:
It is indeed a duplicate of https://github.com/Ultimaker/Cura/issues/9039
That's why I will close this issue. You can find a workaround there and continue the thread if necessary.

closed 2021-01-25 by fvrmr