Ultimaker/Cura #9169: Print Sequence - One at a Time - Suggestion

teknokraze, 2021-01-23:
**Is your feature request related to a problem? Please describe.**
Multiple users have been frustrated with the defined Gantry height causing issues, or not slicing, while printing multiple models One at a Time and the models are aligned front to back and not side to side.

**Describe the solution you'd like**
I, and I'm assuming others, would like to see a sub-menu item added under "Print Sequence" called "Print Order", with selections called "Front to Back" and "Back to Front" or similar, (I believe Cura already slices one at time prints to print models front to back). I'd like it if this feature would then disable the gantry height comparison and only use the printhead size to determine collision points. Having "Front to Back" or "Back to Front" print order selections would allow the user to decide which side of the print head has more or less clearance and be able to print in the order that allows tighter spacing of models, if necessary. 
-The Hover tool tip for the print order could have a warning that states the user must manually ensure the gantry won't be passing over other models.

fvrmr, 2021-01-25:
Thank you for your feature request. This is a duplicate of https://github.com/Ultimaker/Cura/issues/4835 
Here you can find more information about print order and feel free to continue the thread there.
That's why I will close this issue. 

@Ghostkeeper explains in this thread how the order is decided. 

> Let's say you have a printer with a print head that is a giant 10cm cube, but the nozzle is located in the front left corner. You need to print lots of objects so you want to fill the build plate as much as you can.
> 
> If you start printing an object from the front left corner of your build plate, the second object can be placed directly adjacent to the first. This is fine, because the 10cm cube of the print head is located on the other side of the nozzle, to the back and to the right of your nozzle. You could put the second print either on the right of the first, or more towards the back. Either way, your print head will not collide with the first object. Then the third object can again be placed to the right or to the back of the second, as long as you're consistent.
> 
> If you'd start printing from the back right corner of your build plate, the first object still prints fine. But to print the second object, you'd need to print it >10cm to the left of the first object, because there is a giant print head in the way. Or you'd need to print it >10cm in front of the first object, because there is that 10cm block on your print head.
> 
> Cura encodes the "shape of the print head" in the shadow area around your objects, if you're printing in one-at-a-time mode. It does an operation on the shape of the print head that reduces the area (polygon intersection with its 180 degree rotation) and then devises the order in which it must print in order to be able to print so close together.
> 
> If the order changes, so must the disallowed areas. The shadows around the object can only grow bigger, because the order that Cura devises is optimal to minimise the size of the shadows.

Hope that this makes sense to you.

teknokraze, 2021-01-25:
Umm...no...it's not duplicate. Feature request <a href="https://github.com/Ultimaker/Cura/issues/4835" target="_blank">#4835</a> only comments on print order for One at a Time print sequence. Not gantry height limits. Basically, my gantry for the X axis is 30mm high from the bed. But if the models are higher than 30mm Cura says it can't be sliced. But if I chose a print order that that I know the gantry won't pass over the already printed model there's no reason the file shouldn't slice. This is why selecting a "Print Order" option would/could disable the Z axis gantry limit and allow manual print order selection instead of the Cura generated print order.

fvrmr, 2021-01-26:
Thank you for your explanation, I discuss this with the team. Keep you posted.

konskarm, 2021-01-27:
A similar suggestion about the disabling the Z axis gantry limit when the print order is manually selected, is going on in #4835. So I would argue this is still a duplicate. Please add any further comments there, so that we can keep the "Print Order" feature suggestions in the same thread.

closed 2021-01-27 by konskarm