Ultimaker/Cura #9170: print preview has lost is visual usefulness

scathappens, 2021-01-23:
in previous version of cura the print preview gave you a solid preview in a colour and was very readable, in the new version of cura with all the lines, colours and see through  area it becomes hard to read the model. while i can see the benefits of the new style it would be good to have the option to use either method not be stuck with this first one...i use print preview to diagnose models printing issues and this really doesnt help communicate these issues to other people.

Regards 
Neil

fvrmr, 2021-01-25:
Hi @scathappens, what option do you mean with solid preview? 
Recently we made Line type the default in preview. See https://github.com/Ultimaker/Cura/issues/9154

nallath, 2021-01-25:
It could also be that it's defaulting to the legacy mode of the preview. This is either because you selected it, or because your graphics card reports that it doesn't support openGL 4.1

If you can share logs & a screenshot, that would also tell us more.

closed 2021-02-08 by no-response[bot]