Ultimaker/Cura #9171: Cura keeps resetting my printer settings - Ender 3 V2

eblieb, 2021-01-24:
**Application version**
4.8.0

**Platform**
Window 10 x64 Professional
OS Build 19042.746

**Printer**
Creality Ender 3 V2

**Actual results**
Cura constantly resets my Printer Start G-Code

**Expected results**
Cura save my Printer Start G-Code and not constantly reset it

**Additional information**
I keep changing my Ender 3 V2 Start G-code to change the G28 line to G28 R (since my Z axis tends to fall when the steppers are disengaged, the R moves the Z axis up so that my BLTouch doesn't fail and require a power cycle of the printer.

; Ender 3 Custom Start G-code

G92 E0 ; Reset Extruder

G28 R; Home all axes
 

But Cura 4.8 keeps reset it to

; Ender 3 Custom Start G-code

G92 E0 ; Reset Extruder

G28 ; Home all axes
 

I have no idea why it keeps resetting it. Any ideas?

fvrmr, 2021-01-25:
I'm not able to reproduce this issue. Could you share your log file please?
You can find your log file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

eblieb, 2021-01-25:
[cura.log.1.txt](https://github.com/Ultimaker/Cura/files/5869099/cura.log.1.txt)
Here you go. You can see where it switches from G28 R; to G28

fvrmr, 2021-01-26:
I see that you are loading a project file: `PLA+_SmartTemperatureTower_PLA_Plus_195-235.3mf` that will probably overwrite your printer settings.

eblieb, 2021-01-26:
Any way to prevent a 3mf file from overriding printer settings?

Ghostkeeper, 2021-02-01:
Yeah. In the general preferences there is a dropdown that says "Default behavior when opening a project file". If you set that to "Always import models" the 3MF files will never override printer settings any more (project files essentially just become plain 3MF models). If you set that to "Always ask me this", it'll come up with a pop-up when you load the 3MF file to ask you whether you want to use the settings in the file or not.

In the log files, we can see that the start g-code changes after loading in that project file. We'll assume that's what caused it to reset then.

closed 2021-02-01 by Ghostkeeper