Ultimaker/Cura #9175: Cura 4.8 long startup time (over 30 seconds).

oliof, 2021-01-24:
**Application version**
Cura 4.8.0 (with Startup Optimizer installed and USB printing disabled)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
Ultimaker Original+

**Reproduction steps**
1. Reboot computer
2. Start Cura
3. Wait for main window to appear

**Screenshot(s)**
Screenshot not applicable 

**Actual results**
It takes up to 45 seconds for the main Cura window to show and be usable

**Expected results**
Application startup should be comparable to other slicers (5 seconds for superslicer, 2 seconds for KISSlicer). 

**Project file**
Project file does not apply.

**Log file**

1. [DxDiag.txt](https://github.com/Ultimaker/Cura/files/5862034/DxDiag.txt)
2. [cura.log](https://github.com/Ultimaker/Cura/files/5862040/cura.log)

**Additional information**

Without startup optimizer, the startup time of Cura can approach 2 minutes.
I have confirmed the startup time several times using [PassMark AppTImer](https://www.passmark.com/products/apptimer/).

After startup, Cura performs reasonably.

fieldOfView, 2021-01-25:
Do you have a network drive mapped to drive letter X: that may be unavailable or slow to respond? If so, try changing that driveletter to a different letter.

nallath, 2021-01-25:
> Do you have a network drive mapped to drive letter X: that may be unavailable or slow to respond? If so, try changing that driveletter to a different letter.

I already asked that, but it seems to not be the case. I've also asked if the UM3NetworkPlugin was enabled (but all this was on discord, so not mentioned here yet)

oliof, 2021-01-25:
I do not have a network  drive mounted to letter X

fvrmr, 2021-01-25:
I will discuss this with the team. Keep you posted!

konskarm, 2021-01-26:
We cannot really specify a reason as to why it takes so long for you. In my local machine, without using the startup optimizer or disabling anything, it takes 8-12 seconds to load. Could it be that your harddrive is slow?

oliof, 2021-01-26:
My harddrive is a middle of the road consumer Samsung PM981 SSD at 70% capacity. Since other programs start up within reasonable time frame, I am bound to believe it's not an issue with the hardware. That said, I have a replacement drive on order (to increase local storage space), so I will be able to compare and contrast with an EVO970 in a couple days.

oliof, 2021-01-26:
The Samsung EVO 970 Pro came early. Unfortunately, startup times are wild! I did three runs, and the startup times were

75 seconds
18 seconds
20 seconds

I am willing to assume the 75 second one was an outlier. The 40% startup time improvement compared to the old SSD seen in the second and third runs is about the performance gain the EVO 970 Pro should show compared to the PM981 according to spec sheet and synthetic benchmarks I could find browsing the internet.

To compare with the other slicers I regularily use, KISSlicer 2.0 alpha opens in under one second, Superslicer in about 10, and Simplify3D in about 8 seconds.

nallath, 2021-01-27:
The first startup is always longer since it creates cache files (so that is to be expected).

I do have some improvement ideas to speed up the startup time, but we would need quite a bit of refactoring to get that to work.

Ghostkeeper, 2021-01-27:
I have regular hard disks, moving plates and all, the HGST Ultrastar He10.
Takes about 6 seconds for Cura to start for me. Something is still up with Oliof's installation as far as I can see, not with his hardware (although it could still be slow RAM; most operating systems cache files in RAM for a while once accessed).

From your log file it shows that loading in the plug-ins already takes over 7 seconds, about half of the start-up time. This is pretty much just importing Python code. There is very little we can do to optimise that; just write less code I guess. Although some of the old version upgrade plug-ins are probably not used by you any more (which can shave off half a second of that).

Aside from that it also spends 5 seconds loading in the metadata of all of the built-in profiles. This is something that I'd expect the StartupOptimiser plug-in to reduce. It's hard to say if this is the reduced time and how much effect it's had, but from what I've hard other users say it's quite a significant effect so this would otherwise probably be the largest chunk of the start-up time. We have an idea to improve this on our backlog (another cache level on hard disk).

And then it spends 3 seconds loading the interface. This is again something that will be very hard to optimise. I'm supposing most of that is the creation of QML elements and there will probably be some gains we can find but this requires extensive profiling, followed by refactoring to then reduce the QML element count.

One thing you could try is to disable the Settings Guide, see if that makes a difference, since that has in the past caused slowdowns on start-up. Those slowdowns should be fixed in recent versions by loading the articles in the background after start-up. But looking at your log it doesn't seem to wait completely until the screen is visible. The Settings Guide needs to parse hundreds of Markdown files after all.

oliof, 2021-01-27:
Disabling all the version upgrade plugins brought startup time down to 13.0 to 13.6 seconds. Disabling Settings guide shaved off another half second which I am gladly willing to accept in my life.

Here is a current log file from the new disk and without the version upgrade plugins

* [cura.log](https://github.com/Ultimaker/Cura/files/5879661/cura.log)

(PS: Seems like issue #9139 may be related?)

oliof, 2021-01-27:
And a fresh logfile from a boot with all the version upgrade plugins disabled and the Settings Guide enabled (27.4sec first startup, 12.3sec, 12.3sec subsequent startups according to AppTimer, 23.3sec, 10.6, 10.48 according to the log entries).

[cura.log](https://github.com/Ultimaker/Cura/files/5879752/cura.log)

oliof, 2021-01-27:
Upgraded my laptop from 8GB DDR4 RAM to 2x8GB DDR4 RAM -- no change in start up times. I think this rules out RAM pressure due to availability or RAM speed.

oliof, 2021-01-28:
The only thing that's got terrible performance on my Laptop is the GPU: https://www.userbenchmark.com/UserRun/39118626 and that doesn't come into play much before startup.

Ghostkeeper, 2021-02-01:
Yeah, so the lowest hanging fruit are not really a big influence on start-up time then. It'll be hard to make significant improvements beyond that. A lot of time is spent in parsing the Python code (limitation of working with the programming language we're using) and in building the interface. How hard it'd be to remove interface elements to make that load faster, I don't know, but it'd be the best way to improve this in the long term I think.

I mean, other than switch to C++ and destroy basically Cura's entire ecosystem to date.

oliof, 2021-02-01:
That surely would be too big a request (-: 

There is indeed a long gap between the startup splash and the main window appearing (now 5 to 8 seconds, a good 45 seconds on my original hardware configuration). I guess the poor i5-8250 with 8 threads just can't keep up.

Ghostkeeper, 2021-02-03:
Creating the interface elements is all done on the same thread by Qt, as far as I know. Qt has a "main thread" which is separate from our own main thread in Python and generally all interface updates happen in that one thread. As far as I know there is no way to change that in Qt.

oliof, 2024-01-05:
startup times have masisvely improved in the 5.x series. Closing this.

closed 2024-01-05 by oliof