Ultimaker/Cura #9176: Weight estimation displayed as a decimal, not an integer

evilC, 2021-01-24:
**Is your feature request related to a problem? Please describe.**
Not a functional problem, no

**Describe the solution you'd like**
Predicted weight should show fractions of a gram

**Describe alternatives you've considered**
None

**Affected users and/or printers**
Anybody printing small parts where weight is an issue

**Additional context**
I print a lot of very small parts for small drones, where every fraction of a gram is significant. It would be nice if the predicted weight was shown to at least 1 or 2 decimal places. I guess this data is available, just rounded out in the UI. There is plenty of space for it, so it seems to not be an issue. It looks like it is rounding down, because some of my ~1g parts are showing 0g as the estimation

fvrmr, 2021-01-25:
Thanks for you feature request, I will discuss this with the team. Keep you posted!

closed 2021-01-26 by nallath

mattolan, 2021-04-01:
What was the outcome of this? I would also like to be able to see the grams displayed to the two decimal place.  
0.00 for example

nallath, 2021-04-02:
It was implemented in a commit: https://github.com/Ultimaker/Cura/commit/464c19f5925a8e374f801271336e6f6a25d66909

This means it's fixed for 4.9

evilC, 2021-04-20:
@nallath @mattolan This change was not implemented in 4.9
What was done was that the **length** now shows a decimal, but the **weight** does not
[image]

fvrmr, 2021-04-29:
Hi @evilC if you hover over the info icon. It show the weight with decimals. 
[image]

evilC, 2021-04-29:
Ah, thank you!

commit 464c19f5 (2021-01-26): Show print weight with single decimal in the details view

Fixes #9176
CURA-7992