Ultimaker/Cura #9177: Adaptive infill

jrmoserbaltimore, 2021-01-24:
**Is your feature request related to a problem? Please describe.**
When filling a large part, strength is easily achieved by a denser infill (e.g. 20%) near the outside, but infill throughout is unnecessary.  When using a light infill sufficient for a large part, small, load-bearing connections get no infill, and become weak.

**Describe the solution you'd like**
An adaptive infill would solve this.  For small protrusions or generally narrow areas, a dense infill (e.g. 20% gyroid) would supply strength; broad areas could either begin with a dense infill (e.g. 20% gyroid) and steadily adapt into a light infill (e.g. 5% gyroid) over a gradient, or they could use an inner wall.  I like the idea of a steady gyroid gradient, but that may be more difficult to achieve.

**Describe alternatives you've considered**
Thicker walls don't provide as much structural integrity per unit material as infill or we would use 5% gyroid + thick walls all the time.

Detecting narrow parts and filling them specially is an idea, but it's probably simpler to make a generic adaptive infill and let the problem sort itself out instead of trying to discover when a specific area needs structural integrity.

Manually altering the infill area is a possibility requiring a lot of end user work, as well as both internal infrastructure and UI work.

**Affected users and/or printers**
Anyone printing large, hollow parts needing significant but minimal structural integrity will benefit from immense material savings using this feature.

**Additional context**
[image]

smartavionics, 2021-01-24:
See https://github.com/smartavionics/Cura/releases/tag/20201121 for a Cura derivative that implements the gradient infill feature as described in https://www.cnckitchen.com/blog/gradient-infill-for-3d-prints. Gradient infill simply reduces the infill line width away from the edges of the print rather than modifying the pattern density as you are suggesting.

Asterchades, 2021-01-25:
Cubic Subdivision also already does basically this by default, though it will attempt to never crop a complete cube. It becomes more obvious the greater the infill density or larger the cavity.

fvrmr, 2021-01-25:
Thanks for your feature request. 

You can use the plugin Smart slice: https://www.tetonsim.com/smart-slice-for-cura. You can find it in the marketplace (good to know that this is a paid plugin).
This will probably do the job for you. But I will discuss this with the team, but I think it will be a big change to make it fully automatic. Keep you posted!

fvrmr, 2021-01-26:
I discussed this with the team. But like I said before it is an big and complex change. It will also not work for all infill types. 
And what @Asterchades already mentioned `Cubic Subdivision` does this basically. 
That's why I will defer this issue.

closed 2023-07-13 by github-actions[bot]