Ultimaker/Cura #9179: infill line direction is wrong

JohnCalin, 2021-01-24:
**Application version**
(The version of the application this issue occurs with.)
4.8.0

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Gentoo Linux, Intel integrated gpu

**Printer**
(Which printer was selected in Cura?)
Anycubic Mega X

**Reproduction steps**
1. Slice the attached project.
2. Examine layers 47 and 51
3. Notice that one section on layer 47 and a different section on layer 51 have infill going the wrong direction
4. See the screenshot

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
(What happens after the above steps have been followed.)
The infill on some sections of some layers use a different direction than the rest of the layer

**Expected results**
(What should happen after the above steps have been followed.)
Infill should have a consistent direction throughout each layer

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)
[clear_hollow.zip](https://github.com/Ultimaker/Cura/files/5863053/clear_hollow.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
Log from slicing.  Full execution log can be uploaded if needed.
[cura_slice.log](https://github.com/Ultimaker/Cura/files/5863069/cura_slice.log)

**Additional information**
(Extra information relevant to the issue.)
If the settings in this project look odd, it's because I'm printing a solid shape with a hollow space inside, using transparent filament.
I'm getting pretty decent results, but on this particular model the sections that are printed in the wrong direction end up with air bubbles, and come out opaque.
Because infill is set to 100%, the top/bottom direction setting is used instead of infill direction.  In the uploaded project the setting is set to default, but I have tried customizing it.  The top/bottom direction setting works as expected for most of the model, but the sections described above are unchanged by the setting - they still use a specific angle regardless of the setting.

fvrmr, 2021-01-25:
Can you share a picture or video of where it goes wrong while printing. The infill directions looks at the best way to fill and that means that sometimes the directions change.

JohnCalin, 2021-01-25:
The previously attached image shows that one section of infill uses a different direction than the rest.
The printing doesn't 'go wrong' per se, but I'm trying to accomplish a very specific goal (truly clear solid objects).  This is an unusual use case, so I'm not surprised that the default behavior of Cura is a poor fit.  In other respects Cura is amazingly configurable, easily allowing for a huge number of unusual use-cases to be satisfied. I opened this bug because Cura is using some internal logic to override my explicit settings, without any warning and (as far as I can tell) no mechanism for me to override the internal logic.  This seems contrary to Cura's goal, so I assumed it was a bug.

smartavionics, 2021-01-25:
It's because that skin is being treated as a bridge skin because it spans two islands on layer 46. The earlier bridging code in Cura would align the skin lines so that they point from one island to another. If you enable the experimental bridge settings it may do the right thing or not with Cura 4.8. Certainly if you use my Cura variant (see https://github.com/smartavionics/Cura/releases/tag/20201121) it does it right as far as I can tell...

[image]

JohnCalin, 2021-01-25:
The build you linked above does still have the behavior - layer 46 never had the problem, it was on 47 and 51.
The Enable Bridge Settings option under Experimental DOES resolve it.
Thank you!

smartavionics, 2021-01-25:
> The build you linked above does still have the behavior

Yes, because when the experimental bridge settings are disabled, that build uses the original code that causes this issue.

JohnCalin, 2021-01-25:
Got it, I thought you were suggesting either/or.
The setting solves it in both 4.8 and the build listed above.

closed 2021-01-25 by JohnCalin