Ultimaker/Cura #9184: Simplify the lowest top layers where the top surface is complex

martinsykes1, 2021-01-25:
Consider a textured base for a miniature - the top contains a lot of complex topology but there is no benefit in recreating this internally on the underside of the top surface by rigorously applying the number of top layers. It would be faster to simplify the internal layers so they were continuous.

[image]

In the picture, the top image is the current situation and as it nears the top it spends a lot of time creating complex islands which it tries to print over the infill. In the suggested bottom version, it joins these up to create a continuous surface which will print quicker and more successfully. 

It essentially joins separate polygons where it can do so within the interior of the object

fvrmr, 2021-01-26:
Thank you for your feature request. 
I'm not sure if I get your point. 

Considering your drawing is a side view, you could use `Per object` option in Cura. 
Add a cube or other object in the same model. Select `Modify settings for overlaps`  - `Cutting mesh` you can then select the settings you want to change for where the cube is overlapping your object. 

[image]

As you can see the bottom part is printed with infill 20% and the top part (where the cutting mesh stops) is printed with infill 100%. 

[image]
[image]

Please let me know if this could be a solution.

martinsykes1, 2021-01-26:
Thanks – that looks like it does exactly what I want on a simple object such as my example. Unfortunately, many objects are much more complex and it wouldn’t be practical to do that so if there was some way to get the same or similar result automatically it would be much more useful.

I think if I was coding it, I’d do something simple like making the top layers always start on a layer which was a multiple of 4 for example:

 Top layer Existing Upper layers Expanded top layers                                                                                                                                                                                                                                    

From: fvrmr
Sent: 26 January 2021 13:12
To: Ultimaker/Cura
Cc: martinsykes1; Author
Subject: Re: [Ultimaker/Cura] Simplify the lowest top layers where the topsurface is complex (#9184)

Thank you for your feature request.
I'm not sure if I get your point.
Considering your drawing is a side view, you could use Per object option in Cura.
Add a cube or other object in the same model. Select Modify settings for overlaps - Cutting mesh you can then select the settings you want to change for where the cube is overlapping your object.

As you can see the bottom part is printed with infill 20% and the top part (where the cutting mesh stops) is printed with infill 100%.

Please let me know if this could be a good solution.
—

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

fvrmr, 2021-01-26:
Thanks for your reply. I can't see your attached pictures because you probably replied in an e-mail. Please add them in a comment on Github.

martinsykes1, 2021-01-26:
The solution describes does exactly what I want but is only practical for simple items whereas the problem applies to much more complex items than my original example. 

Without knowing how the code currently works, my suggestion for a simple (i assume) coding approach would be to simply take the lowest surface of the top layers and extend them down to a layer below where the new layer is for example always a multiple of 4. This would have the effect of simplifying the hidden lower layers and increasing print quality and speed without being computationally expensive. 

[image]

fvrmr, 2021-01-26:
Thanks, I will discuss this with the team. But I think it is complex to code. Keep you posted!

konskarm, 2021-01-27:
The most straightforward way to deal with that issue is, as @fvrmr suggested, to use a cutting mesh. Alternatively, a more automated approach can be achieved using the setting `Minimum Infill Area`. Setting that to a high enough value eliminates infill parts that are smaller and replaces them with skin, effectively doing what you request here. Play around with the value of this setting and you will get to what you are looking for.

closed 2021-01-27 by konskarm