Ultimaker/Cura #9185: [4.8.0] Hole horizontal expansion causes incorrect model

zabloudil, 2021-01-25:
**Application version**
4.8.0

**Platform**
Windows 10 
AMD Radeon R5 M330 

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Import a simple 1 mm thick hollow cylinder shape model
2. Set Hole horizontal expansion to 0.3
3. Slice

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

Preview
[image]

**Actual results**
Everything is fine up to the height of the hole. Then the wall is suddenly thinner. 

**Expected results**
The wall has a constant thickness throughout the model.

**Project file**
[CE3_model.zip](https://github.com/Ultimaker/Cura/files/5869242/CE3_model.zip)

fvrmr, 2021-01-26:
Hi @zabloudil, this is expected behaviour in Cura. 
If you set your `Hole horizontal expansion` to 0, the wall has a constant thickness. But I am assuming that you want the whole cylinder being shrunk. 

Let me explain what is happening in Cura. 
Cura sees only the top part (above the square door) as a hole. So that's why that part is being shrunk, according to the hole horizontal expansion. 
The bottom part is a continuous contour, so not really a hole according to Cura. That's why it doesn't get expanded, this results in the different sizes between the bottom part and the top part of the object. 

Layer 75 is not a hole, while layer 76 is a hole. 
[image]
[image]

Please let me know if this makes sense to you.

zabloudil, 2021-01-27:
@fvrmr thank you for explanation. It's clear now but this behavior (what is a hole or not) looks strange to me. I tried another model but the parameter did nothing in this case. Must be a hole closed to take the parameter into account ?
[image]

Block137, 2021-01-31:
_Note that inner compensation only works on CLOSED inner contours. Even though it is almost closed, is counted as an outer contour because it is connected to the outside. So only outer diameter compensation will take effect._

This is an explanation from Chitubox but the same rule is applied here I think.

Source: https://www.chitubox.com/en/article/support/howto/chitubox-free/settings/how-to-use-tolerance-compensation-on-chitubox?fbclid=IwAR2T4PBJ9xtfVS_KpJ-QgPZh3Vmjb0JrFMxoavuIuSNFC57wy4Rtn1NQDao

nallath, 2021-02-02:
> _Note that inner compensation only works on CLOSED inner contours. Even though it is almost closed, is counted as an outer contour because it is connected to the outside. So only outer diameter [...]

Yup. It's only a hole if it's not connected to the outside.

zabloudil, 2021-02-02:
I understand. It's not a bug so this ticked can be closed.  Thank you

closed 2021-02-02 by zabloudil