Ultimaker/Cura #9186: line width not working properly

Sirwin47, 2021-01-26:
**Application version**
4.8

**Platform**
windows 10, Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz   3.70 GHz

**Printer**
ender 5

**Reproduction steps**
1. changed wall line width to 0.26, changed top/bottom line width to 0.26, left line width at .4 and initial layer line width at 100%
2. printed, everything changes to 0.26

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
first layer looks horrible, under extruded, problem with adhesion, not printing at 0.4 width

**Expected results**
fist layer to print at 0.4 width, outer walls and top to print at 0.26mm

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Asterchades, 2021-01-26:
The first layer is invariably going to be comprised entirely of walls and bottom extrusions, which you had told to have a 0.26mm width. The 100% modifier is going to be applied to that 0.26mm, resulting in 0.26mm. In other words, it's doing exactly what you told it to do.

If you want the initial layer to print with 0.4mm extrusions in spite of the static values you've assigned those features, you'll want to use approximately a 153.856% first layer modifier.

Sirwin47, 2021-01-26:
Thanks for clearing that up, I was just wasn’t thinking about it the
proper way.

fvrmr, 2021-01-26:
Thanks @Asterchades for explaining. 

I will remove the bug label and change it to question.

closed 2021-02-04 by Ghostkeeper