Ultimaker/Cura #9187: HELP: Combined infills / Linear advance codependency

studiodyne, 2021-01-26:
**When using  combined infills ,  for ex : Outlines in 0.1mm in 4 passes , and Infills in 0.4mm in one pass  (for gain of time )**

An issue appears , the machine doesn't have the ' linear advance setting for the 0.4mm layer height ' and  each lines in 0.4 are shorter , don't overlap , and infills are SHIT ! . If i use only one linear advance setting , either i have big corners or i have unglued infills...

Question : How to solve ?
Is there a solution in cura to send gcode command to modify linear advance when layer height increase or change ?
Is there a tab script ' layer height change' to place a gcode script

Idea : Easy to fix , just make a PR that send a gcode ' comment' with attached ' current layer height' , and we will scrip this 

Thks to not close THIS SERIOUS POST
Thks

fvrmr, 2021-01-28:
Hi @studiodyne, can you maybe share a project file so I can reproduce this. 
Maybe you can use the post processing script `search and replace` or `change at Z`

Ghostkeeper, 2021-02-04:
Cura does not implement Linear Advance, nor user-defined settings to depend on other settings (the latter can be done when editing the profile .cfg files though). The Search and Replace script can be made to insert commands to change Linear Advance settings at a certain layer, though that is a bit technical.

I suggest you request this as a feature for [the plug-in you're probably using to get a Linear Advance setting](https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/issues), although there is already a similar [feature request](https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/issues/20) there too.

closed 2021-02-04 by Ghostkeeper

studiodyne, 2021-02-05:
Yes it's a Feature request, but I would like to suggest this as a bug as well. Because CURA offers fine contours and strong fills, and that poses a problem. CURA must absolutely inform the MACHINE that it is changing the way it prints. A machine set to 0.1mm 'with of course an LA' setting, will create a problem if CURA prints at 0.4mm.
CURA must enter this change for the machine to apply the appropriate settings. For example, send a comment in the GCODE like 'Current layer height = xxx', and the firmware of the machine applies the new settings

So, for me it's a bug, at least a lack of finish

I would like to stimulate the CURA team to produce an 'information' for the machine when it changes print height, nothing more.

You closed the discussion, but, it would be very nice if you could get the information back, and maybe even change things so that a small fix is ​​applied.

Thanks anyway
Steeve

studiodyne, 2021-02-05:
We discussed and did an expertise on MarlinGit and, there are no real 'serious' solutions, to guess at what height to print. We mentioned the detection of multiple extrusions but it's still 'delicate' or 'poker game', in fact it's almost impossible to code a system that can guess what the pressure is in the nozzle and apply the LA settings. So all you have to do is solve this

studiodyne, 2021-02-05:
Last Message 
I don't want CURA LA implementation
I don't want LA plugin
I HAVE IT in my machine
I just want CURA give me the information when printing with an other layer height than the ORIGINAL

Thkssssssssss again

Ghostkeeper, 2021-02-12:
If the layer height has been adjusted from the default in Cura, it will appear in the bottom of the g-code as an adjusted setting. Here's a snippet of the g-code output at the end of a file generated by Cura 4.8:

```
;SETTING_3 {"global_quality": "[general]\\nversion = 4\\nname = Normal #2\\ndefi
;SETTING_3 nition = ultimaker3\\n\\n[metadata]\\ntype = quality_changes\\nqualit
;SETTING_3 y_type = fast\\nsetting_version = 18\\n\\n[values]\\nlayer_height = 0
;SETTING_3 .16\\n\\n", "extruder_quality": ["[general]\\nversion = 4\\nname = No
;SETTING_3 rmal #2\\ndefinition = ultimaker3\\n\\n[metadata]\\ntype = quality_ch
;SETTING_3 anges\\nquality_type = fast\\nsetting_version = 18\\nposition = 0\\n\
;SETTING_3 \n[values]\\n\\n", "[general]\\nversion = 4\\nname = Normal #2\\ndefi
;SETTING_3 nition = ultimaker3\\n\\n[metadata]\\ntype = quality_changes\\nqualit
;SETTING_3 y_type = fast\\nsetting_version = 18\\nposition = 1\\n\\n[values]\\n\
;SETTING_3 \n"]}
```

It's an encoding of the unsaved setting changes. Halfway through you'll find `layer_height = 0.16`, indicating that I changed the layer height to 0.16mm.

Does that solve your problem?

We are not about to implement a sub-feature of linear advance if we don't have support for linear advance at all.

studiodyne, 2021-02-12:
@Ghostkeeper  
Thank you for this message which finally looks like a message that I wanted to read !!!!!!!!! I'll check in cura to see if these 'tweak comments' apply to every change. And if so, then I'll come and thank you twice