Ultimaker/Cura #9190: Arachne - Spiralize tries to do inner contour as well as outer

tlhintoq, 2021-01-26:
Tried spiralizing a cup.  Just a simple drinking cup.
Got all sorts of extra travels.  Upon zooming in I realize it didn't spiralize just the outer contour.  It tried to process the inner contour as well.  The picture makes it pretty clear what's happening.

[image]

jellespijker, 2021-01-26:
Thank you for your feedback on the Arachne engine. Could you fill in the complete bug report and provide the project file as well?
Log files are also more then welcome but I think less relevant for this bug.

Keep in mind that the spiralize function is still being generated the "old" way. Except for the bottom part.

Therefore spiralize will give you limited insight into the unique features and behavior of the variable line width of Arachne. That being said, due to the major rework of the code around this feature it will still have some hickups. Your report and project file will certainly help us along.

Looking forward to the additional information and keep up the testing.

tlhintoq, 2021-01-27:
Email sent to @jellespijker with .3mf file

Ghostkeeper, 2021-02-04:
@jellespijker Did this issue still reproduce on the current master? I don't recall seeing it in our issue board, but maybe I've forgotten it.

jellespijker, 2021-02-10:
@Ghostkeeper Sorry, this notification got snowed under. I just checked and it is still an issue in the current arachne_rebased. The 3mf file slices 4 cups in on-at-a-time mode. But the problem still occurs with a single model.  
[image]

For some reason both the inner and the outer surfaces of the model are used. This doesn't happen in 4.8. The mesh seems to be okay. Ik will put it up on the eCCB for discussion today.
[image]

jellespijker, 2021-02-12:
Devs added to the backlog for the beta version as ticket CURA-8019

Ramalama2, 2021-06-07:
i can confirm the same
[image]

fvrmr, 2022-02-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

WanaGo, 2022-02-01:
After following #11333, just a comment about this.
Would it not then make more sense for Cura to enable/disable options as is required to make spiralize mode work, automatically? ie when you click the tick box, it will temp disable or enable other options that are required to make it work. I agree the descriptions are not at all clear as to what is actually causing the problem, all we can see is it doesn't work.

Ghostkeeper, 2022-02-03:
Spiralise mode in particular has pretty strict constraints on the types of models you can print with it.

While profiles can make it better, there are no spiralise-profiles in Cura as far as I know. But in general it is infeasible to constrain models to what could be printed with spiralise. We could edit the models in Cura, but it's easy to edit it such that the original intent of the user is lost. And Cura is not a modelling editing tool, either.

closed 2023-06-13 by github-actions[bot]