Ultimaker/Cura #9191: Use parts of the infill as transition for filament changes within one structure

AlexHuebi, 2021-01-26:
This feature request is "relatively" simple. I want to use a different filament for the wall than the infill is as a cost saving measure. But if use different filaments, a prime tower is needed since I don't want (eg black) filament as a part of the wall (that would be wooden filament in my case).

If a bit of the infill can be used as a transition, then the walls could be clean and the prime tower could be turned off.

Using the prime tower is wasteful and cant be used in some cases if the printed part is almost as large as the build area itself. Using only one filament is too expensive for this usecase where a bit of structural integrity is needed. 

One example where a prime tower doesn't even fit:
[image]

fvrmr, 2021-01-28:
Hi @AlexHuebi thank you for your feature request. I will discuss this with the team. Keep you posted!

Ghostkeeper, 2021-01-29:
Duplicate of https://github.com/Ultimaker/Cura/issues/3657.

closed 2021-01-29 by Ghostkeeper