Ultimaker/Cura #9192: Suport Interface Type: Raft

KenLucke, 2021-01-26:
**Is your feature request related to a problem? Please describe.**
Creating a set of parameters to create a support interface on a large flat (horizontal) surface without leaving scars and damage is virtually impossible.  Sanding is an option, but some materials (notably "silk/satin" type PLAs) do not do well will sanding (losing their sheen and brilliance).

**Describe the solution you'd like**
Because a raft provides a smooth support surface for the bottom of a model, yet with proper settings is easily removable, it would be nice to replace the standard "support interface" with a raft-type structure, complete with air gap, etc., that would be built in its place at the top of the underlying support matrix.  Options could include "Minimum support area" before this mode kicked in (the surface would have to be horizontal), plus the remainder of the normal raft settings (or it could inherit those from the raft settings in the Build Plate Adhesion section

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
Just about anyone who prints with supports.

This would enable printing of things such as this with a quality surface on what would eventually be "the top" (yes, this _could_ be printed in three pieces with the top 180° [flat on the bed], but that's not always the case, I just had this model laying about to use as an example):

[image]

fvrmr, 2021-01-28:
Hi @KenLucke thank you for your feature request. I will discuss this with the team. Keep you posted!

fvrmr, 2021-01-29:
I discussed this with the team. We came to the conclusion that if you use `support interface` you can use a lot of settings that are the same in the raft settings. 

Here you see the list of settings (left are the raft settings and on the right are the same settings for support)
`Build Plate Adhesion Extruder` = `Support Roof Extruder`
`Raft Extra Margin` = `Support Roof Horizontal Expansion` (you also need to increase the `Support Horizontal Expansion`)
`Raft Air Gap` = `Support Top Distance`
`Raft Top Layers` = `Roof Thickness`
`Raft Top Line Width` = `Support Roof Line Width`
`Raft Top Spacing` = `Support Roof Line Distance`
`Raft Top Print Speed` = `Support Roof Speed`
`Raft Top Print Acceleration` = `Support Roof Acceleration`
`Raft Top Print Jerk` = `Support Roof Jerk`

Only the settings `Raft Smoothing`, `Intintial Layer Z Overlap`, `Raft Top Layer Thickness` and `Raft Fan Speed` doesn't exist in the settings for support. 

Please let me know if you this will work for you.

KenLucke, 2021-01-29:
I've tried fiddling with all those settings already, and I never can get as smooth and removable an interface surface as I get when I use a raft.  Oh, well....  Thank you anyway.

Ghostkeeper, 2021-01-29:
It should be pretty much identical as the raft if those settings are equal to the raft equivalents.

KenLucke, 2021-01-29:
I know it should.  But _should_ and _is_ are not always the same :).

Let me try it some more.  I have prints going now that are multi-day prints, so it will be a few days until I can try it again.  I'll put this on hold until then.

closed 2021-02-26 by no-response[bot]