Ultimaker/Cura #9193: Default scale settings per printer

evertdespiegeleer, 2021-01-27:
**Is your feature request related to a problem? Please describe.**
The motor settings on the printer I'm using are well tuned. The finest possible adjustment however is (obviously) 1 step/mm. For small prints, this usually provides pretty accurate prints. However, when working with prints approaching the size of the build plate, the errors accumulate to about 2mm for me. There's no point in increasing or decreasing the number of steps per mm. Both a step more or a step less decrease the dimensional accuracy; this is the best I can get by adjusting the step rate.

There is however another degree of freedom in the equation, being the number of millimeters.
I usually narrow down the remaining error by scaling the model and this way increasing the dimensional accuracy for big prints.
The scaling factors I use are obviously very close to 1 and are determined experimentally.
These factors only depend on the hardware and the motor settings, so they don't aren't related to the part geometry and don't ever change.

This solution works well, but it is clearly a bit of a stupid thing to do manually for every single print.

**Describe the solution you'd like**
I think it would be really useful to be able to configure default scaling factors for every printer in the printer settings. These could then automatically apply depending on the selected printer, in the same way the build plate dimensions adjust when selecting another printer.

**Affected users and/or printers**
The problem I described is assumably less important on printers that use lead screws on all axes, because of the high gear ratio. My printer has a lead screw for the Z-axis, but uses timing belts with pulleys on the motors for X and Y.
Because of this, the number of steps per mm on my Z-axis is about 5 times as high as on the X- and Y-axes, giving me much finer control over Z in the motor settings.

fvrmr, 2021-01-28:
Hi @evertdespiegeleer thank you for your feature request. I will discuss this with the team. Keep you posted.

fvrmr, 2021-01-29:
We will not do this feature request because we think this is really specific per person and we believe not many users will use this. That's why I will label this with won't fix/do. 
 
However there is a work around that may work for you. You can add `material_shrinkage_percentage` in your printer profile. You can find the .cfg or .def.json in `definitions` or `definitions_changes`: `C:\Users\username\AppData\Roaming\cura\cura version`  
This will shrink your object when you slice it.
Please let me know if that works for you.

Ghostkeeper, 2021-02-04:
We also discussed that we really already implemented something like this with `material_shrinkage_percentage`. The setting wasn't entirely ready for release due to the user experience of what happens with the collision areas in the GUI, and because experimentally the scale factor didn't seem to fix the shrinkage issue well. So it was hidden from the GUI for now. But we do plan to fix the material shrinkage at some point by compensating for it.

closed 2023-07-13 by github-actions[bot]