Ultimaker/Cura #9196: Support not working

Masters50, 2021-01-28:
I am using CURA 4.8.0 and when I have the support box checked no support is added to my print. I tried adjusting it from 0-90 degrees and nothing shows. Even the red marks that show where support is needed doesn't show up. This was working and now it doesn't.

fvrmr, 2021-01-28:
Hi @Masters50 I need some more information to look into this. Can you share your project file? 
Is in preview the checkbox helpers enabled?

Masters50, 2021-01-28:
> CAUTION: This email originated from outside of the LTC organization. Do not click links or open attachments unless you recognize the sender. CAUTION
>
> Hi @Masters50 [github.com]<https://urldefense.com/v3/__https:/github.com/Masters50__;!!FpElZmRfiOuRtA!R2vLHD9LXGU3NZ3HV1wHADWmXK0KlgGYjfIQek-NX6ANSmwOhQt_mGcfhXjBfdtHFMP1irA$> I need some more information to look into this. Can you share your project file?

Good morning,

I finally gave up and just uninstalled Cura and then reinstalled it. It seems to be working now but when I was trying it out, I noticed something that seems unusual to me. I have attached my file and another picture. I realize I should flip this part over to print it and no support would be necessary. But I was using this part and its placement as a test. I remember the first version of Cura that I had installed would show red in areas that support was recommended. Now I don’t see that.

Thank you,
Steve

fvrmr, 2021-01-28:
I can't see your attached picture and file, because you replied in an email. 

But I guess that in your preferences `display overhang` is disabled. 
[image]

If you still want me to look at your file and picture can you add them in a .zip in a comment on Github?

Masters50, 2021-01-28:
Hi,
Display overhang is not disabled…I’m busy at work now, but I’ll post something on Github later. Thanks for your time !

[Screenshot (30)][user-images.githubusercontent.com]<https://urldefense.com/v3/__https:/user-images.githubusercontent.com/77000924/106154116-8ed50c80-617f-11eb-99dd-77148c9350e6.png__;!!FpElZmRfiOuRtA!Wztb74LetRXrhPL6ZBeD_00U6PE5vRQXvPV5_WNNaabXzx2z1tyebOzk1tRhG_HtUn7AMcM$>

Masters50, 2021-01-29:
I finally gave up and just uninstalled Cura and then reinstalled it. It seems to be working now but when I was trying it out, I noticed something that seems unusual to me. I have attached my file and another picture. I realize I should flip this part over to print it and no support would be necessary. But I was using this part and its placement as a test. I remember the first version of Cura that I had installed would show red in areas that support was recommended. Now I don’t see that.

Thank you,
Steve
[support issues.zip](https://github.com/Ultimaker/Cura/files/5890902/support.issues.zip)

fvrmr, 2021-01-29:
Hi @Masters50, thank you for your files. 
It is normal to see no red in preview. You can only see that in the prepare view. 

For the support try to change the `support x/y distance` to 0.

Please reach out if you need more help or have more questions, but I will close this issue because it is not bug related.

closed 2021-01-29 by fvrmr