Ultimaker/Cura #9197: Retraction speed cannot be increased above 50mm/s with the SV-01 printer profile

Splic3dBread, 2021-01-28:
**Application version**
4.8.0

**Platform**
Windows 10

**Printer**
Sovol SV-01

**Reproduction steps**
Using my printer's default profile, the retraction speed cannot be increased beyond 50mm/s without an error. I tested out a cr-10 and ender-3 profile on the same installation and didn't run into this problem.

**Screenshot(s)**
[image]

**Actual results**
Increasing the retraction speed beyond the default value results in an error.

**Expected results**
Increasing the retraction speed beyond the default value should not result in an error / failure to slice

**Project file**
n/a

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5884380/cura.log)

**Additional information**
n/a

fvrmr, 2021-01-28:
Hi @Splic3dBread this is not a bug.
This is related to your printer definitions. We don't maintain the printer profiles for this printer. But you can change it yourself and even make a pull request if you want, so others can use it too. 

You can change the `maximum_value` for `retraction_speed` in this file: `SV01.def.json`
You can find that file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\definitions`

Please let me know if you need help or have more questions.

Ghostkeeper, 2021-02-04:
The maximum value is by default the limit imposed by your firmware. Indeed it can't be increased beyond 50mm/s if the movement speed limit of the firmware is 50mm/s.

Ghostkeeper, 2021-02-04:
This is not deferred because it's not an issue of priority. It's just an issue of that we can't change profiles for 3rd party printers since we don't have those printers.

closed 2021-02-04 by Ghostkeeper

Oderik, 2021-11-16:
Is the maximum speed actually read from the firmware of the printer (if it's connected via OctoPrint) or is it just a configuration value that eventually reflects a firmware default?

Ghostkeeper, 2021-11-24:
It's a configuration value. The people that make the printer definition files are expected to align it with what the firmware's default is.
People who edit their firmware can also edit it in the .def.json files or using the Printer Settings plug-in.