Ultimaker/Cura #9198: Red warning for travel speed >149mm/s

alexbonf, 2021-01-28:
**Application version**
4.8.0

**Platform**
Windows 10 - GPU Nvidia RTX 2070 max-Q

**Printer**
Tronxy X5SA Pro

**Reproduction steps**
1. Import an STL File
2. Set a travel speed >149mm/s

**Screenshot(s)**
[image]

**Actual results**
Red warning in travel speed

**Expected results**
warning should be >500mm/s or so

**Project file**
Doesn't happen on a specific file

**Log file**
No crash involved. However, let me know if required

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-01-28:
Hi @alexbonf this is not a bug. 

This is related to your printer definitions. We don't maintain the printer profiles for this printer. But you can change it yourself and even make a pull request if you want, so others can use it too.

You can change the `maximum_value` for `speed_trave`l in this file: `tronxy_x5sa.def.json`
You can find that file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\definitions`

Please let me know if you need help or have more questions.

Ghostkeeper, 2021-01-29:
By default the maximum value is the value that Cura knows for your printer's firmware (sqrt(2) * maximum feedrate X/Y). So even if you want to increase it, your printer firmware is probably not going to allow you print that fast. That's why Cura colours it red.

The Maximum Feedrate can also be adjusted with the Printer Settings plug-in from the marketplace. Just know that your printer will probably not go any faster for it and time estimates will be further off then.

closed 2021-02-02 by nallath

alexbonf, 2021-02-02:
Hello,

Sorry for the late reply. 

@fvrmr . Thanks for your answer. I am trying to modify those settings but in the folder you have mentioned there is no file. Am I missing something? However, I did find a file named tronxy_x5sa.def.json which is in the installation folder (D drive in my case). Still no speed_travel parameter in there but such a file inherits tronxy_x.def.json so I tried to look at it. There I do see speed_travel but I do not see any maximum value. Below is what I see:

"speed_travel": { "value": "60.0 if speed_print < 50 else 120.0 if speed_print > 80 else speed_print * 1.25" },

Am going through the correct path or am I doing something completely wrong? 

@Ghostkeeper, thanks for your answer as well. However, I do not understand how the feedrate is related to travel speed. To the best of my knowledge, the feedrate determines the volume of filament being extruded per second. Therefore travel motion should be performed by different motors during non-extrusion actions. Could you please explain better?

Thanks both!

Ghostkeeper, 2021-02-04:
Feedrate has little to do with the feeder. It's a bit of a misnomer in that regard. The term "feedrate" comes from CNC milling. See also https://en.wikipedia.org/wiki/Speeds_and_feeds . I think it has something to do with how fast the material is "fed" to the cutter.

In CNC, "feedrate" refers to the movement speed of any axis (X, Y, Z or E). It is determined by the `F` parameter in the g-code commands.

In my opinion it is easier to download the Printer Settings plug-in from the marketplace and edit those settings in Cura's interface than to go digging through the files. However if you did want to edit the `maximum_value` for `speed_travel` in that file you'd need to define it as a separate property alongside the current `value` property, like this:

```
"speed_travel": {
    "value": "60.0 if speed_print < 50 else 120.0 if speed_print > 80 else speed_print * 1.25",
    "maximum_value": "999",
}
```

Note that you'd need to write a similar `maximum_value` property for all speed settings that you want to increase beyond the firmware's capabilities. And it would probably still not print that fast on the actual printer unless you modify the firmware as well.