Ultimaker/Cura #9200: Crash when loading recent project

BagelOrb, 2021-01-28:
**Application version**
Current master

**Platform**
WSL 2

**Printer**
UM S5

**Reproduction steps**
1. Save as project
2. Close
3. Start Cura
4. File > Open Recent > ...curaproject.3mf

**Actual results**
Crash

**Expected results**
No crash.

**Logs**
```
Traceback (most recent call last):
  File "/mnt/c/Development/CuraApp/Cura/cura/CuraApplication.py", line 1765, in readLocalFile
    workspace_handler.readLocalFile(file, add_to_recent_files = add_to_recent_files)
TypeError: readLocalFile() got an unexpected keyword argument 'add_to_recent_files'
```

```
Thread 0x00007f29127fc700 (most recent call first):
  File "/mnt/c/Development/CuraApp/Cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/usr/lib/python3.8/threading.py", line 870 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f2912ffd700 (most recent call first):
  File "/usr/lib/python3.8/threading.py", line 306 in wait
  File "/usr/local/lib/python3.8/dist-packages/zeroconf/__init__.py", line 2530 in wait
  File "/usr/local/lib/python3.8/dist-packages/zeroconf/__init__.py", line 1735 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f29137fe700 (most recent call first):
  File "/usr/lib/python3.8/threading.py", line 306 in wait
  File "/usr/lib/python3.8/threading.py", line 558 in wait
  File "/mnt/c/Development/CuraApp/Cura/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "/usr/lib/python3.8/threading.py", line 870 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f2913fff700 (most recent call first):
  File "/usr/lib/python3.8/threading.py", line 306 in wait
  File "/usr/local/lib/python3.8/dist-packages/zeroconf/__init__.py", line 1485 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f2924e91700 (most recent call first):
  File "/usr/local/lib/python3.8/dist-packages/zeroconf/__init__.py", line 1359 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f2925692700 (most recent call first):
  File "/mnt/c/Development/CuraApp/Cura/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/usr/lib/python3.8/threading.py", line 870 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f293a7fc700 (most recent call first):
  File "/usr/lib/python3.8/threading.py", line 302 in wait
  File "/usr/lib/python3.8/threading.py", line 433 in acquire
  File "/mnt/c/Development/CuraApp/Uranium/UM/JobQueue.py", line 98 in _nextJob
  File "/mnt/c/Development/CuraApp/Uranium/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f293affd700 (most recent call first):
  File "/usr/lib/python3.8/threading.py", line 302 in wait
  File "/usr/lib/python3.8/threading.py", line 433 in acquire
  File "/mnt/c/Development/CuraApp/Uranium/UM/JobQueue.py", line 98 in _nextJob
  File "/mnt/c/Development/CuraApp/Uranium/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f293b7fe700 (most recent call first):
  File "/usr/lib/python3.8/threading.py", line 302 in wait
  File "/usr/lib/python3.8/threading.py", line 433 in acquire
  File "/mnt/c/Development/CuraApp/Uranium/UM/JobQueue.py", line 98 in _nextJob
  File "/mnt/c/Development/CuraApp/Uranium/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f293bfff700 (most recent call first):
  File "/usr/lib/python3.8/threading.py", line 302 in wait
  File "/usr/lib/python3.8/threading.py", line 433 in acquire
  File "/mnt/c/Development/CuraApp/Uranium/UM/JobQueue.py", line 98 in _nextJob
  File "/mnt/c/Development/CuraApp/Uranium/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Current thread 0x00007f299ac08740 (most recent call first):
  File "/mnt/c/Development/CuraApp/Cura/cura/CrashHandler.py", line 396 in _logInfoWidget
  File "/mnt/c/Development/CuraApp/Cura/cura/CrashHandler.py", line 182 in _createDialog
  File "/mnt/c/Development/CuraApp/Cura/cura/CrashHandler.py", line 86 in __init__
  File "cura_app.py", line 167 in exceptHook
  File "/mnt/c/Development/CuraApp/Uranium/UM/Qt/QtApplication.py", line 383 in exec_
  File "/mnt/c/Development/CuraApp/Cura/cura/CuraApplication.py", line 852 in run
  File "cura_app.py", line 241 in <module>
```

**Additional information**
[image]

fvrmr, 2021-01-29:
Hi @BagelOrb this is probably fixed with https://github.com/Ultimaker/Cura/commit/9197a1ca51d56312aa027dcfa80375022486ed15
Please me know if you have this crash after you tried it with this fix.

BagelOrb, 2021-02-01:
I'll take your word on this one.

closed 2021-02-01 by BagelOrb