Ultimaker/Cura #9201: Infill bone structure

matzejo, 2021-01-29:
Is it possible to implement something like an infill pattern inspired by bone structures? 
Infill Optimization for Additive Manufacturing - Approaching Bone-like Porous
Structures
https://www.google.com/url?sa=t&source=web&rct=j&url=http://orbit.dtu.dk/ws/files/128479821/WuAagSig17.pdf&ved=2ahUKEwiy_b_26MDuAhVx7OAKHUrBCh0QFjAAegQIAxAB&usg=AOvVaw3owByWmdTcxLKrWJjevFWV

fvrmr, 2021-01-29:
Hi @matzejo thank you for your feature request.
Can you explain why you want an infill pattern inspired by bone structures? 
And could you add the picture in a comment, I can't open the link.

matzejo, 2021-01-29:
Shure :) [image]

You can download the whole pdf on orbit.dtu.dk 
[image]

Asterchades, 2021-01-29:
Here's the actual link to the PDF the OP was referring to, without all the Google link-bloating/-tracking stuff: https://orbit.dtu.dk/files/128479821/WuAagSig17.pdf

It's fortunately considerably higher resolution than the image provided.

matzejo, 2021-01-29:
A bone structure is optimised for rigid parts without using to much material. The gradual filling steps are fine but they work primary for top and bottom layers. There is a lag on the side wall support. Maybe a kind of cellular structure will also work with big bubbles in the middle and small bubbles close to the walls.

fvrmr, 2021-02-01:
Thanks, I will discuss it with the team. Keep you posted!

fvrmr, 2021-02-02:
We discussed it with the team. We all think it is very cool. 
But we also see that it is very complex and would take a lot of time to implement. So that why I will defer this issue.

Matszwe02, 2021-04-13:
I saw something similar, it can be implemented it the "bone structure infill" wouldn't be implemented.
https://www.youtube.com/watch?v=hq53gsYREHU

Ghostkeeper, 2021-05-03:
@Matszwe02 see also https://github.com/Ultimaker/Cura/issues/6954

Matszwe02, 2023-06-29:
> Is it possible to implement something like an infill pattern inspired by bone structures? Infill Optimization for Additive Manufacturing - Approaching Bone-like Porous Structures [...]

Hi. As github bot assigned `stale` in this feature request, maybe creating this request also in PrusaSlicer will be ok? I think it's a great idea to enhance infill, so let more developers know of this enhancement.