Ultimaker/Cura #9206: Print Speed Parameter in Print settings window not same with Print Seeds in Prefernece - Profile Window

tommybj79, 2021-01-29:
**Application version**
(The version of the application this issue occurs with.)
4.8.0

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 on Core i7 vPro with Intel HD

**Printer**
(Which printer was selected in Cura?)
Anycubic i3 Mega -> Mega Pro

**Reproduction steps**
1. I change the Print SPeed at the Print Setting window and save the changes to current profile
2. (Something you did next.)

**Screenshot(s)**

[image]

**Actual results**
At preference window their is no change of the print speed. I also try restart of Cura

**Expected results**
(What should happen after the above steps have been followed.)
The print speed seeting at the print setting -porfile section, should be same with the seeting I see at the preference window

fvrmr, 2021-02-01:
Hi @tommybj79, you are looking at the global settings. Did you check Extruder 1 settings?

Ghostkeeper, 2021-02-05:
Duplicate of https://github.com/Ultimaker/Cura/issues/7626.

closed 2021-02-05 by Ghostkeeper