Ultimaker/Cura #9209: Print order

Julius971, 2021-01-30:
The idea is to make a print procedure that goes like:
1. Innermost wall
2. infill
3. rest of walls as defined (out to in or in to out)

The idea is, or i hope that this could hide some extrusion imperfektions (caused by oozing retract problems) to the inside of the print. The idea with the innermost wall is, to give the infill some outer points to stick to and to minimise visibility from the outside.

fvrmr, 2021-02-01:
Hi @Julius971 thank you for your feature request. 
Can you give an example why this will work better?

Julius971, 2021-02-01:
This could improve surface quality in situations where nouzle bleeding is happening. If you wont have your retraction settings on point, you might start overextruded or under extruded at the beginning of the line after a longer travel move.
This order would simply place the effects of this extrusion fails to non visible areas. When the nozzle starts printing the outer walls, those imperfections should have been gone and left on the inside of the print.

nallath, 2021-02-02:
> This could improve surface quality in situations where nouzle bleeding is happening. If you wont have your retraction settings on point, you might start overextruded or under extruded at the beginning of the line after a longer travel move.

But that is also the case if you print it like this:
1. Inner walls
2. Outer walls
3. Infill.

Julius971, 2021-02-02:
Partially, but the smaller the segment gets, the more the error would fade towards the outermost wall. 
I mean, i know my idea wouldn't be "the" solution. But i think it could be a minor thing to improve the print by another bit.

Ghostkeeper, 2021-02-08:
I think the flow switch from infill to outer wall would cause a much greater error then.

Julius971, 2021-02-08:
Assuming you use different flowrates.

Ghostkeeper, 2021-02-08:
Indeed. I'm saying this because most profiles have their infill print much faster than their inner walls or outer wall.

Julius971, 2021-02-08:
Yea, but for lin_advance users where pressure and speed becomes more and more irrelevant, this wouldn't affect it anymore.

Ghostkeeper, 2021-02-09:
Right, but then the oozing would also be less of a problem if Linear Advance is working properly.

As an alternative, the "Extra Infill Wall Count" setting adds a wall specifically around your infill which gets printed along with your infill. If you use the same flow rate for infill and walls, or if flow rates are indeed irrelevant, then this prints precisely in the order that you ask.
The intention of Extra Infill Wall Count is to allow the infill lines to rest on something at the edge. Otherwise it could not connect to skin very well. That is sort of a related goal to catching blips, since if the lines don't rest well on their endpoints then that could lead to stringing and blips due to the lines being dragged along.

Julius971, 2021-02-09:
Ok the extra infill wall makes the deal. Didn't know about that. 👍🏻

closed 2021-02-09 by Julius971