Ultimaker/Cura #9211: Connect enhancement for USB printer

rgb55, 2021-01-31:
Hello, this could be a nice enhancement for USB printer.

Whit USB connected printer this take very long to reconnect the printer if Cura is open and the printer vas turned of to avoid printer fan noise and then turned on again.
A good idea would be to have a connect button to reconnect easily the printer without restarting Cura or wait for a long time that the printer is connected back to Cura.

Regards,
Roland

fvrmr, 2021-02-01:
Hi @rgb55 thank you for your feature request. 
USB connection is not a focus of Ultimaker, because Ultimaker printers uses network connection. That's why we won't look into this and why I will defer this issue. 
But a pull request is always welcome, so you and other users that have USB connection can use it.

rgb55, 2021-02-01:
Thanks for your feeback,

 

Roland

Ghostkeeper, 2021-02-08:
A manual connect option is being discussed in this feature request: https://github.com/Ultimaker/Cura/issues/5207

closed 2021-02-08 by Ghostkeeper