Ultimaker/Cura #9212: [4.8.0] Top half of support missing

mrmichaelrb, 2021-01-31:
**Application version**
4.8.0

**Platform**
Windows 10 64-bit

**Printer**
SeeMeCNC Rostock Max V2.0

**Reproduction steps**
1. Sliced a model.

**Screenshot(s)**
[image]

**Actual results**
The included screenshot shows where support for a part of the model is generated up to a certain layer and then stops.

**Expected results**
The support should continue up to meet the bottom edge of that part of the model.

**Project file**
[FlySky FS-TH9X Monitor Mount.zip](https://github.com/Ultimaker/Cura/files/5899821/FlySky.FS-TH9X.Monitor.Mount.zip)

**Log file**
[cura.zip](https://github.com/Ultimaker/Cura/files/5899822/cura.zip)

**Additional information**
If I rotate the model around the Z axis (I tried 45 degrees), I can get the support to generate correctly.

I've also included the STL model used and the GCODE generated:
[FlySky FS-TH9X Monitor Mount STL and GCODE.zip](https://github.com/Ultimaker/Cura/files/5899824/FlySky.FS-TH9X.Monitor.Mount.STL.and.GCODE.zip)

fvrmr, 2021-02-01:
Hi @mrmichaelrb
If you set your `support join distance` to 1 or lower it will solve your problem. 

[image]

Please let me know if it also works for you. 

Maybe you can also try our Arachne engine alpha:
https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_alpha
I sliced your model with it and the corner parts looks a lot better because of the variable line widths.

mrmichaelrb, 2021-02-01:
> If you set your `support join distance` to 1 or lower it will solve your problem.
> Please let me know if it also works for you.

Thanks @fvrmr, it solved my problem.

> Maybe you can also try our Arachne engine alpha:
> https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_alpha
> I sliced your model with it and the corner parts looks a lot better because of the variable line widths.

I've just switched all my slicing from Simplify3D (which has gone stagnant) to Cura, with noticeable improvements (especially on small features and overhangs with experimental settings used). As soon as things settle a little, I'll try out the Arachne engine.

fvrmr, 2021-02-02:
Good to hear! I will close then since this is solved.

closed 2021-02-02 by fvrmr

Ghostkeeper, 2021-02-08:
I wonder why having too high of a Support Join Distance would cause this support to disappear though?

mrmichaelrb, 2021-02-09:
Logically I would say it seems like a bug. I don't see any reason why it should happen. But my problem is solved, and it seems like they are moving on to a new Arachne engine that I assume does not exhibit this problem, based on the test slice that @fvrmr did.