Ultimaker/Cura #9216: [4.8] Unit test "code-style" failing with multiple "has no attribute" errors

onitake, 2021-01-31:
**Application version**
4.8

**Platform**
Debian GNU/Linux (bullseye/sid)

**Printer**
N/A

**Reproduction steps**
1. Run the unit tests from the build tree: `/usr/bin/ctest --force-new-ctest-process -j4`

**Screenshot(s)**
See https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=980603 for the full output.

**Actual results**
The "code-style" unit test fails with lots of "has no attribute" errors.
Here is a short excerpt:
```
> 10/14 Test #12: pytest-VersionUpgrade44to45 ......   Passed    0.29 sec
>       Start 14: shortcut-keys
> 11/14 Test #14: shortcut-keys ....................   Passed    0.29 sec
> 12/14 Test #13: pytest-XmlMaterialProfile ........   Passed    1.10 sec
> 13/14 Test  #3: pytest-main ......................   Passed   30.22 sec
> 14/14 Test  #1: code-style .......................***Failed   61.38 sec
> cura/PrinterOutput/Models/MaterialOutputModel.py:6: error: Module 'PyQt5.QtCore' has no attribute 'pyqtProperty'
> cura/Settings/SettingVisibilityPreset.py:6: error: Module 'PyQt5.QtCore' has no attribute 'pyqtProperty'
> cura/Settings/SettingVisibilityPreset.py:44: error: "pyqtSignal" has no attribute "emit"
> cura/Settings/SettingVisibilityPreset.py:49: error: "pyqtSignal" has no attribute "emit"
> cura/Settings/SettingVisibilityPreset.py:54: error: "pyqtSignal" has no attribute "emit"
> cura/Settings/SettingVisibilityPreset.py:59: error: "pyqtSignal" has no attribute "emit"
> cura/API/ConnectionStatus.py:3: error: Module 'PyQt5.QtCore' has no attribute 'pyqtProperty'
```

**Expected results**
All unit tests succeed.

**Project file**
N/A

**Log file**
Full build output: http://qa-logs.debian.net/2021/01/20/cura_4.8-1_unstable.log

**Additional information**
Qt version: 5.15.2

fvrmr, 2021-02-01:
Hi @onitake thank you. We will look into this. Keep you posted.

Ghostkeeper, 2021-02-02:
I'm not getting these, and neither is our CI server (e.g. in the latest push: https://github.com/Ultimaker/Cura/runs/1805124034).
It's also quite illogical. `PyQt5.QtCore` does have `pyqtProperty`, and `pyqtSignal` does have an `emit` method.

I've seen something like this before though, and then it was because of the `PyQt5-stubs` package. That package provides type hints for PyQt5, which makes MyPy check more things. That would be a good thing, except that PyQt5-stubs is frequently outdated. In your case certainly, because their latest release is for PyQt 5.14.2 and your version is more modern. I wouldn't think it needs to break on that but for me it does. The stubs project is kind of on a low fire, judging by [this](https://github.com/python-qt-tools/PyQt5-stubs/issues/115) and [this](https://github.com/python-qt-tools/PyQt5-stubs/issues/23).

On our build server and on my home computer I don't have PyQt5-stubs installed so it never even checks these things (due to the `--ignore-missing-imports` option we give to MyPy).

I'd suspect that uninstalling those stubs makes the errors go away, like it did for me. If not, let me know.
I'll close this ticket since as far as we know, we're calling every PyQt5 thing the correct way in our code. The stubs may be outdated or just plain incomplete. I don't know why it'd need to complain about those attributes otherwise.

closed 2021-02-02 by Ghostkeeper

Ghostkeeper, 2021-02-08:
It still catches quite a few mistakes in our code from time to time, just through our automated CI runs. It's not checking everything, but if it checks most of our own code that already has value for us.

We do already add the `--ignore-missing-imports` flag for MyPy:

https://github.com/Ultimaker/Cura/blob/79ec71ca4c14a2fdb62656e6bc0844734b4b35b1/run_mypy.py#L69-L72

The real PyQt modules don't have typing. They are generated with SIP, which doesn't add them. They do a runtime actual check against the type though, which crashes Cura if the type is unexpected, unless we catch that TypeError.

BryceBeagle, 2021-02-22:
> That would be a good thing, except that PyQt5-stubs is frequently outdated. In your case certainly, because their latest release is for PyQt 5.14.2 and your version is more modern. I wouldn't think [...]

Hey guys, I noticed that you mentioned the PyQt5-stubs release issue, so I thought I'd stop by and let you know we (finally) made [a new release](https://pypi.org/project/PyQt5-stubs/5.15.2.0/). Hopefully the project stays up-to-date going forward :+1:. The old maintainer relinquished the project and we've been (slowly) transitioning it to new management.

Ghostkeeper, 2021-03-02:
Oh, thanks for the update and for maintaining those stubs then, BryceBeagle!