Ultimaker/Cura #9217: [4.8] Appstream data is not up to date (404 on screenshot link and others)

onitake, 2021-01-31:
**Application version**
4.8

**Platform**
Debian GNU/Linux (bullseye/sid)

**Printer**
N/A

**Reproduction steps**
Automatic appstream data verification done by the Debian build system.

**Screenshot(s)**
N/A

**Actual results**
Full report here: https://appstream.debian.org/sid/main/issues/cura.html

You may want to fix at least the incorrect screenshot link, as this will affect everything that uses appstream data.

```
Warnings

    asv-cid-desktopapp-is-not-rdns
    cura.appdata.xml:4 - cura.desktop
    The component ID is not a reverse domain-name. Please update the ID to avoid future issues and be compatible with all AppStream implementations. You may also consider to update the name of the accompanying .desktop file to follow the latest version of the Desktop-Entry specification and use a rDNS name for it as well. In any case, do not forget to mention the new desktop-entry in a <launchable/> tag for this component to keep the application launchable from software centers and the .desktop file data associated with the metainfo data.
    asv-screenshot-no-media
    cura.appdata.xml:27 -
    A screenshot must contain at least one image or video in order to be useful. Please add an <image/> to it.
    screenshot-download-error
    Error while downloading screenshot from 'http://software.ultimaker.com/Cura.png': Failed to retrieve 'http://software.ultimaker.com/Cura.png' (HTTP 404: Not Found)
    This might be a temporary server issue.
    gettext-data-not-found
    Unable to find any translation data for specified Gettext domain 'Cura'.
    no-translation-statistics
    We could not extract any translation statitics, althout metadata hints were given that the data exists. Please check the <translation/> MetaInfo tag for spelling mistakes. It also makes sense to ensure all locale files are placed in the right directories (e.g. gettext .mo files in /usr/share/locale/*/LC_MESSAGES/
```

**Expected results**
No errors or warnings reported.

**Project file**
N/A

**Log file**
N/A

**Additional information**
N/A

fvrmr, 2021-02-01:
Hi @onitake thank you. We will also look into this. Keep you posted.

evtrados, 2021-02-02:
Hi @onitake,

I don't think it is a matter of not being up-to-date, but a matter of behavior on different Linux distributions. Our current build server runs CentOS v6 and AppStream does not complain. Same goes for my local CentOS v8 installation. I've only noticed this behavior on Debian-based distributions, since it happened for me on both Ubuntu (18.04 LTS and 20.04 LTS) and Debian buster. Hence, I have a way to solve it ;) :
1. Rename the `<cura-build repository>/packaging/cura.appdata.xml` to `com.ultimaker.cura.appdata.xml`.
2. Change its contents to add the path to the image: 
`<image>https://github.com/Ultimaker/Cura/blob/master/screenshot.png</image>`
3. In line 62 of the `<cura-build repository>/packaging/linux.cmake` file change the `cura.appdata.xml` entry to `com.ultimaker.cura.appdata.xml`.

Please check if this works for you and in the meantime we will check whether the above steps will work on our build server as well. If that's the case we can update the repository to list the new file names and image URL. If not, I'm afraid we have to leave it this way until we upgrade our system (most probably we'll switch to Debian in the future).

onitake, 2021-02-02:
Well... the image link produces a 404, so I *would* consider it outdated.

On the other hand, these are all warnings, so probably not very critical.
But if they can be fixed without too much trouble, they probably should.

onitake, 2021-02-02:
So, I tested this and `https://github.com/Ultimaker/Cura/blob/master/screenshot.png` won't work, because Github returns a website. To get only the image, one should use `https://raw.githubusercontent.com/Ultimaker/Cura/master/screenshot.png`.

Renaming the appdata and fixing the image spec is not enough to make it validate however, the desktop file must be renamed as well.

So, I did the following changes:

1. Rename cura.appdata.xml to com.ultimaker.cura.appdata.xml
2. Rename cura.desktop.in to com.ultimaker.cura.desktop.in
3. Update com.ultimaker.cura.appdata.xml to use com.ultimaker.cura.desktop
4. Update CMakeLists.txt to use com.ultimaker.cura.appdata.xml , com.ultimaker.cura.desktop and com.ultimaker.cura.desktop.in
5. Replace the screenshot tag in com.ultimaker.cura.appdata.xml with:
```xml
  <screenshots>
    <screenshot type="default">
      <image width="1695" height="1002">https://raw.githubusercontent.com/Ultimaker/Cura/master/screenshot.png</image>
    </screenshot>
  </screenshots>
```

With these changes, the resulting metadata passes the validator:
```shell
$ appstreamcli validate com.ultimaker.cura.appdata.xml 

Validation was successful: pedantic: 1

```

I don't know where you found <cura-build repository>/packaging though, this is certainly not in the Cura source tree. Debian builds Cura using the cmake scripts.

I prepared a PR in #9229 , please review. Thank you!

evtrados, 2021-02-03:
Yes, it seems that all is fine on our build server as well after these changes, so we're going to commit them in the coming days.
Did the above steps help you out with regards to the warnings or did you encounter anything new?

evtrados, 2021-02-04:
Sorry, for some reason I could only see your first response when checking yesterday.
Thanks for the PR! We reviewed it and we'll merge it to see whether everything builds properly on our build server.

> I don't know where you found `<cura-build repository>/packaging` though, this is certainly not in the Cura source tree. Debian builds Cura directly using the cmake scripts.
I had in mind that you were using the whole cura-build repository when trying to achieve this.

closed 2021-02-04 by nallath

onitake, 2021-02-04:
Ah sorry, now I get it. You were referring to https://github.com/Ultimaker/cura-build

I'm not familiar with the build repo, we've used the standard Debian build infrastructure since we started packaging Cura 2.5.
There's all kinds of automatic checks these days, and I recently noticed that there were Appstream warnings. That's why I reported them here.

Unfortunately, I don't know exactly how https://appstream.debian.org/ works, so I just verified with the `appstreamcli` utility as documented in the Debian wiki. I think it's part of the checks that appstream.debian.org runs, so we should be good now. The other warnings about gettext and translation statistics don't seem that relevant to me. I might look into them some other time.

onitake, 2021-02-04:
Thanks for merging!

PR #9229 (2021-02-04): Fix upstream metadata
This PR attempts to fix the metadata issues reported by the Debian metadata validator: https://appstream.debian.org/sid/main/issues/cura.html

Fixes: #9217