Ultimaker/Cura #9218: Merged objects have Top/Bottom layers at seam instead of infill

RFBomb, 2021-02-01:
**Is your feature request related to a problem? Please describe.**
I'm not sure if this is a 'feature request' or a bug. Either way, I think it ought to be atleast considered.

I experimented with PrusaSlicer (by the way, I think Cura's settings interface is way better), but I was surprised to see that merged models there intuitively calculated the infill, compared to what Cura did by default for the same model. Allow me to clarify with photos: (Note that I did chance the infill patterns to make the switch to different extruder more obvious in the sliced photo)

Below is layers 33 and 34 of the model using the basic dual-color setup. (Import models -> assign extruders -> merge -> slice):
As you can see, there is full top and bottom layers where you would expect infill to occur (since the object has been merged). 
[image]
[image]

Here is the same part set up and sliced with comparable settings in PrusaSlicer. Layer 34. 
(Process was import 1 model, then click on it and select 'add model' and import second model. This merges the two together.)

[image]

As you can see in the above photo, it just continued on with the infill. 

I was able to get the same effect in cura after a minute or two of messing around. But to get this I had to ungroup the models, move them away from each other to be able to select them individually, assign custom per-model settings to each of them, then change the values, and finally re-merge them. (In this case the bottom object required custom setting for # of top layers to 0, and top object required setting # of bottom layers to 0).
[image]

I have not tested it yet, but I fear that settings those settings up worked favorably in this Cube, since everything lined up. But if the bottom part were to have an exposed shelf, then the shelf would have no top layers from this workaround. 

**Describe the solution you'd like**
Update the 'merged' logic to use intuitive infill rather than having to do a workaround as described above. 

**Describe alternatives you've considered**
Prusaslicer apparently does a good job of it. But I really dislike everything else about the interface. 

**Affected users and/or printers**
Anyone using 'Merged' objects in their print.

**Additional context**
I'd offer to help fix, but I really don't know enough about how the merged object is being handled in the code, nor do I want to potentially screw up evaluating the objects (since that type of logic / scripting is out of my element)

fvrmr, 2021-02-01:
Hi @RFBomb thank you for your feature request. 
Could you share your project file so I can look into this. 

Merging is the same as grouping, but the difference between them is that you use merging specific for dual color prints. Like this globe. 
[image]

To select the water of the globe you can use `ctrl click` on the water part of the globe. And then you can select one part in a merged object. 

Did you already looked into cutting mesh? Maybe this will also do the job.

RFBomb, 2021-02-02:
I have not looked into cutting mesh, please point me to a resource so I can read up on it if one is available. 

Here are some of the calibration cubes I tested with:
https://www.thingiverse.com/thing:4734209

The cube I used in my pictures though were these two files that were merged together. Nozzle 1 was the lower half, and nozzle 2 was set to the upper half. 
[NozzleOffset_CalibrationCube_Upper.zip](https://github.com/Ultimaker/Cura/files/5911974/NozzleOffset_CalibrationCube_Upper.zip)

Also, thanks for the tip on the ctrl+click. I'll try that out on the next dual-print i set up.

Ghostkeeper, 2021-02-08:
> Also, thanks for the tip on the ctrl+click. I'll try that out on the next dual-print i set up.

I consider this to be an ongoing UX problem. It's hard to discover such hotkeys in Cura :(

The cutting mesh still makes it behave as if it's two separate meshes, by design. This would have the same effect, so it wouldn't solve your issue.

I think the result should not depend on whether two meshes are grouped or not. Nothing else in Cura currently depends on that other than the front-end transformations and one-at-a-time printing. For the front-end transformations it's logical, but for one-at-a-time printing it's not. It would be good to not make that problem worse.

We also can't really apply this always though. There are legitimate cases where you'd want there to be skin between the models, like when you have supports or moulds that are meant to be removed in post-processing. So this would need to be a setting then. My biggest concern is then whether users would understand what the setting does, but I think it can be made to work.

RFBomb, 2021-02-20:
I understand that this is likely quite a complex problem, so I'm not expecting a fix any time soon, especially any change to the algorithm has the possibility of adversely affecting stuff working properly at the moment. 

That being said, I was under the impression that the 'Union Overlapping Volumes' toggle inside of 'Mesh Fixes' is what this was meant to solve. I was playing around trying to see if there was anything in the settings currently when I noticed that toggle. But I was unable to get it to make any noticeable difference in the preview. 

For example, here are some screenshots. 
Layer 31 (Option Checked)
[image]

Layer 31 (Option NOT Checked)
[image]

Layer 32 (Option Checked)
[image]

Layer 32 (Option NOT Checked)
[image]

How Prusa combined that example (for reference):
[image]

Based on this test, and looking at the tool path shown in the preview, the only significant difference between Cura and PrusaSlicer here is that Cura appears to not be calculating the overlapping volumes as infill. Instead, cura weaves 100% infill layers together. The actual edges of the part intersection appears comparable to me, which is a good thing as far as looking for the issue goes.

So from what I can tell the symptoms are:
- Parts joined at a seam will have full # of Top/Bottom Layers for each part at the seam
- Parts with overlapping volumes alternate which extruder fills in the overlapped volume. But this overlap will be at 100% infill. 
- "Union Overlapping Volumes" toggle appears to do nothing, which I think may be the root cause of the problem.

Ghostkeeper, 2021-03-01:
The Union Overlapping Volumes is somewhat unrelated. It applies to the fill rule of a single model that overlaps itself. See the article in the Settings Guide about it: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/meshfix/meshfix_union_all.md

Ghostkeeper, 2021-03-01:
Disabling the "Alternate Mesh Removal" setting causes Cura to behave like what PrusaSlicer does by default. Cura weaves them together by default for extra strength, but this then causes extra printing time.

Cura can also completely ignore the overlap and produce your "whoops" case by disabling the "Remove Mesh Intersection" setting.

RFBomb, 2021-03-01:
Ok, so all that is working as expected then. Thanks for the explanations. The top/bottom layer intersection I still think could be looked at and put onto a toggle like these other settings, but as its not very critical (most if not all of the time it will be hidden), and its there for adhesion purposes between the two objects, then Its really a non-critical fix I think. It'd be nice to have the option for larger models, but I get why its not there. Thanks for the help.

Ghostkeeper, 2021-06-08:
Developers, see issue CURA-8203.