Ultimaker/Cura #9219: 4.8 Modifiers not working when Print Sequence is set to One at a Time

PoiMechanic, 2021-02-01:
**Application version**
(The version of the application this issue occurs with.)
4.6, 4.7, and 4.8
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
MacOS Catalina AMD Radeon Pro 5600M 8 GB
Intel UHD Graphics 630 1536 MB
**Printer**
(Which printer was selected in Cura?)
Prusa mk3s
**Reproduction steps**
1. (Something you did.) Model with a modifier (infill mesh, for overlaps) on it will print fine if the Print Sequence is set to One at a Time but as soon as you click All at Once the modifier settings no longer work and it shows in the preview.
2. (Something you did next.) I've tried stock profiles and everything else with no luck. Tried reinstalling and even going to older versions. Happens with any model or modifier settings.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
In the attached pictures the modifier is puck shaped and on the bottom with a 100 percent infill setting.

**Actual results**
(What happens after the above steps have been followed.) Prints as if modifiers are not there. 
**Expected results**
(What should happen after the above steps have been followed.)
I need to print multiple objects that are the same and have the same modifiers but one at a time. This works fine in Prusaslicer but I looooove Cura

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues
[cura.log](https://github.com/Ultimaker/Cura/files/5901917/cura.log)
 to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
[image]
[image]

[cura.log](https://github.com/Ultimaker/Cura/files/5901935/cura.log)

(Extra information relevant to the issue.)

fvrmr, 2021-02-01:
Hi @PoiMechanic thank you for your report. 
If you group your models(the object and the modifier) it won't happen and it will print them one at a time. 

But I will discuss it with the team. Keep you posted.

PoiMechanic, 2021-02-02:
@fvrmr Thank you so very much. I never knew about that option. This helps me with another problem too. Life saver

Ghostkeeper, 2021-02-15:
Developers, please refer to issue CURA-8031.

nallath, 2022-02-17:
This was fixed in https://github.com/Ultimaker/Cura/pull/11494

closed 2022-02-17 by nallath