Ultimaker/Cura #9220: [4.8.0] Segmentation Fault

danfoster, 2021-02-01:
This feels similar to #8453 , but since this is a different version, I thought it best to open a new issue. 

**Application version**
4.8.0

**Platform**
Ubuntu 20.04.02
Window Manager / DE: i3 using Xorg
GPU: Intel Corporation UHD Graphics 620

(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
N/A - Application crash

**Reproduction steps**
Launch the AppImage

**Actual results**
Shows the loading small dialog box, gets to Loading UI.
Starts to render a new window (I assume the main window), but nothing in it
Then application seg faults

**Expected results**
Application should launch and be usable

**Project file**
N/A 

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5903393/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/5903394/stderr.log)

**Additional information**

Cura used to run fine on this machine. So trying to track down what's changed.

I checked `/var/log/apt/history.log.*` for any auto updates that look relevant, but nothing stood out. For the sake of completeness here is the last few days anyway:

```
Start-Date: 2021-01-28  06:46:12
Commandline: /usr/bin/unattended-upgrade
Upgrade: librbd1:amd64 (15.2.7-0ubuntu0.20.04.1, 15.2.7-0ubuntu0.20.04.2), librados2:amd64 (15.2.7-0ubuntu0.20.04.1, 15.2.7-0ubuntu0.20.04.2)
End-Date: 2021-01-28  06:46:14

Start-Date: 2021-01-29  09:02:54
Commandline: apt install build-essential cmake vim-nox python3-dev
Requested-By: dfoster (1000)
Install: librhash0:amd64 (1.3.9-1, automatic), cmake-data:amd64 (3.16.3-1ubuntu1, automatic), cmake:amd64 (3.16.3-1ubuntu1), libjsoncpp1:amd64 (1.7.4-3.1ubuntu2, automatic)
End-Date: 2021-01-29  09:03:30

Start-Date: 2021-01-29  14:30:59
Commandline: apt install ssh-askpass
Requested-By: dfoster (1000)
Install: ssh-askpass:amd64 (1:1.2.4.1-10)
End-Date: 2021-01-29  14:31:02

Start-Date: 2021-01-29  18:37:38
Commandline: apt install siggen
Requested-By: dfoster (1000)
Install: siggen:amd64 (2.3.10-9), osspd:amd64 (1.3.2-11, automatic), osspd-pulseaudio:amd64 (1.3.2-11, automatic)
End-Date: 2021-01-29  18:37:46

Start-Date: 2021-01-30  21:14:20
Commandline: apt install minicom
Requested-By: dfoster (1000)
Install: lrzsz:amd64 (0.12.21-10, automatic), minicom:amd64 (2.7.1-1.1)
End-Date: 2021-01-30  21:14:47

Start-Date: 2021-01-31  01:07:54
Commandline: apt install xscreensaver
Requested-By: dfoster (1000)
Install: libglade2-0:amd64 (1:2.6.4-2ubuntu2, automatic), libjpeg-turbo-progs:amd64 (2.0.3-0ubuntu1.20.04.1, automatic), xscreensaver-data:amd64 (5.42+dfsg1-1ubuntu1, automatic), xscreensaver:amd64 (5.42+dfsg1-1ubuntu1)
End-Date: 2021-01-31  01:07:59

Start-Date: 2021-01-31  06:48:35
Commandline: /usr/bin/unattended-upgrade
Upgrade: tzdata:amd64 (2020f-0ubuntu0.20.04.1, 2021a-0ubuntu0.20.04)
End-Date: 2021-01-31  06:48:38

Start-Date: 2021-01-31  19:41:29
Commandline: apt install brightnessctl
Requested-By: dfoster (1000)
Install: brightnessctl:amd64 (0.5.1-2), brightness-udev:amd64 (0.5.1-2, automatic)
End-Date: 2021-01-31  19:41:56
```

I've also removed `~/.local/share/cura`,  `~/.config/cura` and `~/.cache/qtshadercache` in the hope of starting from a clean slate, but it did not help.

I tried launching using `env -u DESKTOP_SESSION -u KDE_FULL_SESSION -u XDG_CURRENT_DESKTOP ~/Applications/Ultimaker_Cura-4.8.0_6782abb3abec599bf7274297cdaa94bb.AppImage ` , but it did not help

I tried setting the following in cura.cfg, with no success:
```
[view]
opengl_version_detect = force_legacy
force_empty_shader_cache = True
```

I had enabled the intel xorg driver "TearFree" option (https://manpages.debian.org/buster/xserver-xorg-video-intel/intel.4.en.html) recently, and wondered if that was related, but after commenting that out of my Xorg config and restarting, it still doesn't work.

I can't think of any other changes I've made recently, but there must be something. Could you take a look at the logs above and see if it gives any hints?

fvrmr, 2021-02-01:
Thank you @danfoster for your detailed bug report!!
I will discuss this with the team. Keep you posted!

Ghostkeeper, 2021-02-02:
So according to the log, it's getting a segfault on this line:

https://github.com/Ultimaker/Uranium/blob/4.8/UM/Qt/QtRenderer.py#L134

I looked around the internet for a similar situation but couldn't really find anyone with a similar crash in `glClear`. There are a couple of crashes, but those turned out to ultimately be derived from things like texture buffers which are not used by Cura, or the OpenGL context being closed (but you're getting this at start-up).

We don't really have any way to reproduce this crash at the moment. I think we can only hope to find more people that had the same issue to see what is similar.

danfoster, 2021-02-02:
Thanks for taking a look! I understand how difficult it can be to diagnose a problem when you cannot reproduce it.

I will try to get Cura running from Source (https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Ubuntu). This will hopefully aid in 2 ways:
 1) It will rule out any issues with libraries built and rolled into the AppImage.
 2) Would give us a platform to insert any debugging statements.

danfoster, 2021-02-02:
Got to the bottom of it! It was actually an incompatibility with my Xorg config and not Cura itself. I will describe the issue in the hope it's useful to other end-users encountering the same issue.

As briefly mentioned in my original description, I had recently added the Intel `TearFree` option by creating ` /etc/X11/xorg.conf.d/20-intel.conf` with the following:
```
Section "Device"
   Identifier  "Card0"
   Driver      "intel"
   Option      "TearFree"    "true"
EndSection
```
When I said I commented it out during my debugging, I just commented out the single `Option      "TearFree"    "true"` line, thinking the rest of the `Section` would be harmless. It also changed the default setting for `AccelMethod` from `uxa` to `sna`. Which is the source of Cura crashing.

Adding it back in explicitly, so I ended up with with the following, solved the issue:
```
Section "Device"
   Identifier  "Card0"
   Driver      "intel"
   Option      "AccelMethod"  "uxa"
   Option      "TearFree"    "true"
EndSection
```

For interest, it was clearing the `GL_COLOR_BUFFER_BIT` that caused the segfault, I could clear `GL_DEPTH_BUFFER_BIT` without a crash with `uxa`.

Thanks for your help!

closed 2021-02-02 by danfoster

Ghostkeeper, 2021-02-08:
Is it expected that an application would crash clearing the `GL_COLOR_BUFFER_BIT` when using `uxa`? Seems a bit of a weird acceleration option if the colour buffer can't be cleared! But thanks for finding a workaround! That will be useful for people trying to fix this issue later. I'll modify the title to help that cause.