Ultimaker/Cura #9221: print settings in separate window

gr3333, 2021-02-02:
**Is your feature request related to a problem? Please describe.**
I have a second monitor and it would be really nice to have the print settings window separate, not within the main window. That way, I could move it to my second monitor.

**Describe the solution you'd like**
Have the Print settings 'window'/dialog as a separate window to move aside or to another monitor.

**Describe alternatives you've considered**
Leave it the way it is, but that can limit the view of the current model.

**Affected users and/or printers**
Users with wide or high def monitors or second monitors would benefit from this change.

**Additional context**
N/A

fvrmr, 2021-02-02:
Hi @gr3333 thank you for your feature request. I will discuss this with the team. Keep you posted!

Ghostkeeper, 2021-02-03:
This was attempted before a long time ago but found to be difficult. We're reconsidering though because it was investigated just by one person who wasn't very confident in their QML skills.

fieldOfView, 2021-02-19:
The SidebarGUI plugin will soon let you undock the settings panel into a separate window. Please help testing this feature by downloading this [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/SidebarGUIPlugin/PPQrrZsMNX9X4R2o62qpBhDRIZukamKCv-Z25AUmMSog/SidebarGUIPlugin-v7.0.0-2021-02-19T16%3A00%3A42Z.curapackage?Expires=1929110446&GoogleAccessId=cura-packages-api-production%40stardust-193112.iam.gserviceaccount.com&Signature=XVxOGt9VGMzQnLah8Wx5Z7FNlNusY5HmZLKxoYkiMX%2BA8LtqiYt5hEsjVI%2BdGalu0yRf%2BIjMteC4VxL6mWVJF%2BB%2FbV47JehITEuj3KGLGoHxmtOG8ZcI54wgoHynMhLi3UtBrgchD0jRREM797YEfvBr0oU5bPHMbLZEZ4uBwLaHiVKbtC9sePoxfRq90GjiY%2B3LklMhWSeeJyw6zOMg1nKd7U%2FIQQ5Fsn3ip55ZTGZ0P1Y%2B9wfk%2Fvwh%2Bxh3yeCxqXd20ioNU4bHLoJyFYWnC4x32uYXQHqIPYmTSqCg2TDG34rge2TDYSxj%2Bv2M8OV9CLJYzw75Z6jUP1OXe9I3Vg%3D%3D) and dropping the file onto the buildplate in Cura as if you were opening a 3d model.

The settings can be undocked with this innocuous little button:
[image]

gr3333, 2021-02-22:
Thanks, finally got a few minutes to have a look.
This is **much** cleaner and works well for me.

Ghostkeeper, 2021-02-26:
Works well on Linux too by itself. Except that I can click on the X button of the window to close the separate panel, and then cannot get the settings list back any more. Pressing the X button of the interface works fine, but pressing the X button of the Gnome window itself doesn't seem to register that the window is closed so I can't get the settings back.

fieldOfView, 2021-02-26:
On windows the toolpanel titlebar does not have a close button, like I intended it, so this issue does not exist. I'll see what I can do about this.

gr3333, 2021-06-30:
I really liked this feature and had hoped it would be incorporated into future versions, but I recently downloaded 4.9.1 and see that it no longer works. Am I missing something? Is there a way to implement this again?
Thanks.

fieldOfView, 2021-06-30:
The sidebarguinplugin needs to be updated for *every* new version of Cura.