Ultimaker/Cura #9222: Please help me Translate Cura to vietnamese

Haiang, 2021-02-02:
Currently I am Vietnamese cura from English to Vietnamese, I have overwritten 4 .mo files into the folder.
... / resources / i18n / <language code> / LC_MESSAGES
Then I got the following error"

 

Traceback (most recent call last):

File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 1221, in getSceneBoundingBoxString

KeyError: 'chiều rộng'

 

Thread 0x00002434 (most recent call first):

File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread

File "threading.py", line 862 in run

File "threading.py", line 914 in _bootstrap_inner

File "threading.py", line 882 in _bootstrap

 

Thread 0x000017bc (most recent call first):

File "threading.py", line 297 in wait

File "zeroconf\__init__.py", line 2117 in wait

File "zeroconf\__init__.py", line 1434 in run

File "threading.py", line 914 in _bootstrap_inner

File "threading.py", line 882 in _bootstrap

 

Thread 0x00002fcc (most recent call first):

File "threading.py", line 297 in wait

File "threading.py", line 549 in wait

File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests

File "threading.py", line 862 in run

File "threading.py", line 914 in _bootstrap_inner

File "threading.py", line 882 in _bootstrap

 

Thread 0x00002718 (most recent call first):

File "threading.py", line 297 in wait

File "zeroconf\__init__.py", line 2117 in wait

File "zeroconf\__init__.py", line 1262 in run

File "threading.py", line 914 in _bootstrap_inner

File "threading.py", line 882 in _bootstrap

 

Thread 0x00000a88 (most recent call first):

File "zeroconf\__init__.py", line 1183 in run

File "threading.py", line 914 in _bootstrap_inner

File "threading.py", line 882 in _bootstrap

 

Thread 0x00001388 (most recent call first):

File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread

File "threading.py", line 862 in run

File "threading.py", line 914 in _bootstrap_inner

File "threading.py", line 882 in _bootstrap

 

Thread 0x00001e50 (most recent call first):

File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 165 in _storeStderrToLogThread

File "threading.py", line 862 in run

File "threading.py", line 914 in _bootstrap_inner

File "threading.py", line 882 in _bootstrap

 

Thread 0x00002538 (most recent call first):

File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 153 in _storeOutputToLogThread

File "threading.py", line 862 in run

File "threading.py", line 914 in _bootstrap_inner

File "threading.py", line 882 in _bootstrap

 

Thread 0x000027d0 (most recent call first):

File "threading.py", line 293 in wait

File "threading.py", line 424 in acquire

File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob

File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run

File "threading.py", line 914 in _bootstrap_inner

File "threading.py", line 882 in _bootstrap

 

Thread 0x00002ec0 (most recent call first):

File "threading.py", line 293 in wait

File "threading.py", line 424 in acquire

File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob

File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run

File "threading.py", line 914 in _bootstrap_inner

File "threading.py", line 882 in _bootstrap

 

Thread 0x00002b2c (most recent call first):

File "threading.py", line 293 in wait

File "threading.py", line 424 in acquire

File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob

File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run

File "threading.py", line 914 in _bootstrap_inner

File "threading.py", line 882 in _bootstrap

 

Thread 0x00000cb4 (most recent call first):

File "threading.py", line 293 in wait

File "threading.py", line 424 in acquire

File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 98 in _nextJob

File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 123 in run

File "threading.py", line 914 in _bootstrap_inner

File "threading.py", line 882 in _bootstrap

 

Current thread 0x00000cac (most recent call first):

File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 396 in _logInfoWidget

File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 182 in _createDialog

File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 86 in __init__

File "X:\4.8-exe\build\inst\bin\cura_app.py", line 167 in exceptHook

File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 302 in initializeEngine

File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 912 in runWithGUI

File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 829 in run

File "X:\4.8-exe\build\inst\bin\cura_app.py", line 235 in <module>

File "Console.py", line 24 in <module>

File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible

File "<frozen importlib._bootstrap>", line 664 in _load_unlocked

File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked

File "<frozen importlib._bootstrap>", line 968 in _find_and_load

File "__startup__.py", line 12 in <module>

File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible

File "<frozen importlib._bootstrap>", line 664 in _load_unlocked

File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked

File "<frozen importlib._bootstrap>", line 968 in _find_and_load

fvrmr, 2021-02-02:
Hi @Haiang you can't translate the `width`, `depth` and `height` because these are variable names and must not be translated.

fieldOfView, 2021-02-02:
In general, don't translate any text between `%(` and `)`.

closed 2021-02-02 by fvrmr

Haiang, 2021-02-02:
> In general, don't translate any text between `%(` and `)`.

So the errors in this section I have to leave are not translated, right?
[image]
[image]

fieldOfView, 2021-02-02:
Correct

Haiang, 2021-02-02:
> Correct

Thank you I will try again if there is anything hope you help me more?

Ghostkeeper, 2021-02-09:
We'll try to help out! Most of it is straightforward, but with the plural cases and the replacement keys a lot of translators have had problems.