Ultimaker/Cura #9223: [4.8.0] Cannot run on Void Linux

MaxVerevkin, 2021-02-02:
**Application version**
4.8.0 - AppImage

**Platform**
Void Linux with musl libc

**Printer**
N/A

**Reproduction steps**
1.  Download Ultimaker_Cura-4.8.0.AppImage
2. `chmod +x Ultimaker_Cura-4.8.0.AppImage`
3. `./Ultimaker_Cura-4.8.0.AppImage`

**Screenshot(s)**
N/A

**Actual results**
```
❯ ./Ultimaker_Cura-4.8.0.AppImage
zsh: no such file or directory: ./Ultimaker_Cura-4.8.0.AppImage
```

**Expected results**
Program runs

**Project file**
N/A

**Log file**
N/A

**Additional information**
```
❯ ldd Ultimaker_Cura-4.8.0.AppImage
	/lib64/ld-linux-x86-64.so.2 (0x7f4f91b56000)
	libdl.so.2 => /lib64/ld-linux-x86-64.so.2 (0x7f4f91b56000)
	libpthread.so.0 => /lib64/ld-linux-x86-64.so.2 (0x7f4f91b56000)
	libz.so.1 => /lib/libz.so.1 (0x7f4f9193f000)
	libc.so.6 => /lib64/ld-linux-x86-64.so.2 (0x7f4f91b56000)
Error relocating Ultimaker_Cura-4.8.0.AppImage: gnu_dev_makedev: symbol not found
```

nallath, 2021-02-02:
This might be an upstream issue: https://github.com/AppImage/AppImageKit/issues/1015

MaxVerevkin, 2021-02-02:
I will try to use `libgcompat` then (if I understand the upstream issue right)

MaxVerevkin, 2021-02-02:
Well, I have installed `gcompat` but now I have different issue:
```
❯ ./Ultimaker_Cura-4.8.0.AppImage
This doesn't look like a squashfs image.

Cannot mount AppImage, please check your FUSE setup.
You might still be able to extract the contents of this AppImage
if you run it with the --appimage-extract option.
See https://github.com/AppImage/AppImageKit/wiki/FUSE
for more information
open dir error: No such file or directory

```
I have `fuse` installed in my system.

Saijin-Naib, 2021-07-08:
> Well, I have installed `gcompat` but now I have different issue:

> I have `fuse` installed in my system.
Same behavior with gcompat on Alpine with FUSE installed. Watching what evolves here.

probonopd, 2022-05-13:
fyi, we are currently experimenting with a statically linked AppImage runtime that is known to run on Alpine Linux (= the AppImage gets mounted and the payload application is started). But for the Cura AppImage to run on Alpine, probably bundling additional Cura dependencies would be needed.

* https://github.com/AppImage/AppImageKit/issues/877#issuecomment-1120195107

codewinch, 2022-10-20:
Is this possibly related to Musl on Void and Alpine, so static will resolve these two issues? It's feasible that Void (which has Cura in its repos) on glibc won't have this issue.

MariMakes, 2023-10-19:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, and we've introduced a lot of changes since this report. I'll be closing this issue. I hope you understand.

You can download the latest version of Cura with a bunch of improvements and a new QT version here:
[https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)

Apologies for any inconvenience caused by our delayed response.
Thanks again, and please let us know if you run into any other issues 💪

closed 2023-10-19 by MariMakes