Ultimaker/Cura #9224: Bridging settings affect overhangs(not bridges)

Julius971, 2021-02-02:
**Application version**
4.8

**Platform**
Win 10
specs irrelevant

**Printer**
Ender 5 Pro (customized)

**Reproduction steps**
1. load an overhang testprint
2. enable bridge settings

**Actual results**
If the overhanging parts lines are shorter then the minimum setting for bridges. The overhangs are treated as bridges, not overhangs.

**Expected results**
The bridging settings shouldnt affect overhangs.

fvrmr, 2021-02-04:
Hi @Julius971 could you share a project file? Because I am not sure if I understand what is happening.

Ghostkeeper, 2021-02-09:
I'm unable to reproduce this. Here I have a model of a sheared cube which has a rather steep overhang, steep enough for the entire outer wall to hang in the air.
[image]

Regardless of whether lines are shorter than the Minimum Bridge Wall Length or longer, it's not using the bridging routine for these overhangs. It would print those walls at 50% flow then, and it's not doing that.

A project file would allow us to reproduce your case, so that we can see what's going on.

Julius971, 2021-02-09:
[image] - bridging settings enabled

[image] - bridging settings disabled

Ghostkeeper, 2021-02-11:
Nice pictures!

Julius971, 2021-02-11:
Oh damn missed that. Will do👍🏻

closed 2021-02-26 by no-response[bot]