Ultimaker/Cura #9225: Slicing error with negative horizontal expansion: Cura V4.8.0

swdenman, 2021-02-02:
**Application version**
4.8.0

**Platform**
Windows 10

**Printer**
Creality CR-10S

**Reproduction steps**
1. Open attached object file
2. Set horizontal expansion to -2mm
3. Slice

**Screenshot(s)**
Prepare view:

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-02-02:
Hi @swdenman I can't see your attached files and screenshots. Could you add them?

swdenman, 2021-02-02:
Prepare view:
[image]

Preview (Horizontal expansion 0mm)
[image]

Preview (Horizontal expansion -2mm)
[image]

fvrmr, 2021-02-04:
Thank you @swdenman, this is expected behaviour. Because it applies only on the horizontal axis. And not on the z axis. You can see that in the shadow of your model, the top is not affected by the horizontal expansion. So that's why you get that on your model.
[image]

Hopes that this makes sense to you. Please let me know if you need more 
But I will close this issue now, because it is not a bug.

closed 2021-02-04 by fvrmr

Ghostkeeper, 2021-02-09:
I've made a small modification based on this bug, since the miter limit was also making lines hang in mid-air. It should be a bit better by default in the next version.

swdenman, 2021-02-09:
Thanks for the update.